m5stack/M5Unified #4: ATOM MatrixのIMUだけ方向が違う

tanakamasayuki, 2021-11-01:
M5StickC(or他のボード)とATOM MatrixをUSBポートを手前にして、画面orマトリクスを上にした状態で動かした場合にIMUの数値がATOM Matrixのみ違う
おそらくセンサーの向きが違うので、できれば同じ方向に揃えてほしい

lovyan03, 2021-11-01:
情報ありがとうございます！向きは統一したいですね。

それに加えて、LCDのsetRotationのような感じで、ユーザが向きを設定できる仕組みが用意できないかなと考えています。
とはいえ立体の軸の向きとなると、どういう設定が必要か、どうすれば使いやすいか…要検討ですね。

lovyan03, 2021-11-06:
@tanakamasayuki 
setRotation関数を追加して、developブランチ更新してみました。
examples/HowToUse に使用例を置きました。
```
  // If you use IMU, write this.
  if (M5.Imu.begin())
  {
    if (M5.getBoard() == m5::board_t::board_M5ATOM)
    { // ATOM Matrix's IMU is oriented differently, so change the setting.
      M5.Imu.setRotation(2);
    }
  }
  else
  { // If you use External Unit Accel & Gyro, write this.
    M5.Imu.begin(&M5.Ex_I2C);
  }
```
平面方向の向きをsetRotationにて0～3の4方向で変えられるようになっています。上下方向の設定はありません。
現在はM5.begin時にIMUの有効化をしていないため、ユーザ側コードでの設定になります。
M5.beginの引数は今のところ用意していないのですが、そろそろ実装を考えないといけませんね…。

closed 2021-11-08 by tanakamasayuki