m5stack/M5Unified #5: General questions

patfelst, 2021-11-14:
Hi there @lovyan03 this looks very promising. Is it ready for people to start using?

I have been using @tobozo's [ESP32-Chimera-Core](https://github.com/tobozo/ESP32-Chimera-Core) because I want an up to date TFT_eSPI library for use with M5 Stack Core2. However the 3 touch buttons at the bottom of the LCD were not working properly (.pressedFor(1000) specifically) and tobozo told me about M5Unified.

I tried this library today and found the following issues:
- I couldn't find a way to control the Core2 green LED which is done via the AXP192 "Sys_LED" on GPIO1.
- I use the FastLED library to control a single WS2812 RGB LED with data pin on ESP32 GPIO27. This was working on Chimera-Core but now the LED does not turn on. Not sure what's happening here.
- I have a I2C sensor with SDA on pin 32, SCL pin 33 however I don't seem to be able to communicate with it any more. I initialise it in `Wire.begin(32, 33);`. I then tried `M5.Ex_I2C.begin(I2C_NUM_1, 32, 33);` but this also doesn't seem to work.

I'd appreciate your thoughts / ideas on these issues, thank you.

lovyan03, 2021-11-14:
@patfelst 
Thank you for your interest.
This project is still in its infancy, and I'm sure it will cause a lot of inconvenience, but I hope to make it good.

If you need an external 5V output, you need to call the following function.
```
M5.Power.setExtPower(true);  // If you need external port 5V output.
```

I expect this will probably solve two of your questions.

There is currently no API available for controlling the power LED.
We do not plan to add an LED control function to Axp, as this project needs to work the same way for all models.
I will add the M5.Power.setPowerLED() function instead.

patfelst, 2021-11-14:
Thanks @lovyan03 yes enabling the 5V power output fixed the RGB LED and the I2C sesnor.

For I2C, calling `M5.In_I2C.begin(I2C_NUM_0, 32, 33);` works, however I went back to `Wire.begin(32, 33);`. Again this was 5V power issue.

I look forward to finding M5.Power.setPowerLED() function in a future relase - thanks.

I forgot to say that `BtnX.pressedFor(1000)` is working as I expcted - good!

I will keep testing this library.

patfelst, 2021-11-16:
Actually, I have found an issue with `BtnX.pressedFor()`, but I will raise that in a new issue for this.

All the Lcd functions I have tested are working ok (custom fonts, lines, rectangles).

closed 2021-11-16 by patfelst

robcloot, 2022-04-17:
How do I turn on the Vibration Motor (PWM:AXP_LDO3)?