m5stack/M5Unified #6: Double register of BtnA.pressedFor()

patfelst, 2021-11-16:
Hi there, I'm not sure if this is the correct library to report this, please advise if not.

When `BtnA.pressedFor(1000)` is true, I run some code which takes a few seconds to complete, and in this time, I release touch button A, wait for the Wifi to connect, sync RTC to NTP service. On the next loop, this code executes again, i.e. `BtnA.pressedFor(1000)` is true a second time, even though I am not touching the button.

I found by inserting an extra `M5.update()` the flag is cleared. Can you please advise if this is a bug or not?

```C++
void loop(void) {

  // Update M5 object
  M5.update();

  clock_display.update();

  if (M5.BtnA.pressedFor(1000)) {
    M5.Lcd.clear();
    clock_display.pause();
    // Connect to WiFi
    if (connect_wifi()) {
      delay(500);
      sync_rtc_to_ntp(true);
      WiFi.disconnect(true);  // Disconnect wifi
      WiFi.mode(WIFI_OFF);    // Set the wifi mode to off
    }
    delay(2000); // Give user time to read the Wifi and NTP success / failure message
    init_main_display();
    M5.update();  // Needed otherwise another pressed event occurs on next loop
    clock_display.resume();
  }
  // other code ...
}
```

lovyan03, 2021-11-16:
@patfelst 
thanks for report this.

Well, this behavior is currently as per the spec, but it is also a bug in the spec.
The button state is chatter canceled, and the state change is triggered by staying in the same state for a period of time.
In other words, the first update after releasing the button will still treat the button as being pressed.

M5.update is expected to be called periodically and frequently, so we did not fully anticipate what would happen immediately after a long process.

I will implement a fix. Specifically, chatter cancellation should not be performed if the elapsed time since the last update is more than a certain value.

lovyan03, 2021-11-16:
@patfelst
The develop branch has been fixed. It will be included in the next release.
If you want to try it right away, try to get it from the develop branch.

patfelst, 2021-11-16:
Thanks very much. I will test it now.

patfelst, 2021-11-16:
That fixed it! Thanks for dong it so quickly.

closed 2021-11-16 by patfelst