m5stack/M5Unified #7: What is syntax to create sprite?

patfelst, 2021-11-16:
Apologies for all the issues. What's the preferred way to create a sprite object? Before setup() declare:

`M5Canvas BattSprite;` or
`M5Canvas BattSprite(&M5.Lcd);` or even
`LGFX_Sprite BattSprite;`

They all seem to work. Obviously then in setup():

`BattSprite.createSprite(batt_spr_wdth, batt_spr_ht);` etc

Thank you.

patfelst, 2021-11-16:
Actually they all compile, but the first and last both cuase the ESP32 to crash. The middle one works ok.

patfelst, 2021-11-16:
Thanks. I didn't know you could push a sprite to display in another sprite. I don't need to do it now, but am curious. Would it work like this?
```C++
#include <M5Unified.h>

M5Canvas SpriteA;
M5Canvas SpriteB;

void setup(void)
{
  M5.begin();
  SpriteA.createSprite(32, 32);
  SpriteA.drawCircle(16, 16, 10, TFT_WHITE);

  SpriteB.createSprite(100, 100);
  SpriteA.pushSprite(&SpriteB, 32, 32); // Draw SpriteA inside SpriteB
  SpriteB.pushSprite(&M5.Display, 0, 0);
}

lovyan03, 2021-11-16:
@patfelst 
Yes, that's correct.
If you are interested, take a look at the LovyanGFX sample.

https://github.com/lovyan03/LovyanGFX/blob/0.4.7/examples/Sprite/ClockSample/ClockSample.ino

The entity of M5.Display is the M5GFX library, and M5GFX is a wrapping of LovyanGFX.

patfelst, 2021-11-16:
Wow that clock example will take some study. But I get the idea of what you're doing with the sprites.

closed 2021-11-16 by patfelst