m5stack/M5Unified #10: M5 Stack Core2 crash during create sprite

patfelst, 2021-12-03:
Hi there,

I have been developing on M5 Stack Core2 with this library for a couple of weeks now and all has been good! Yesterday I noticed that I was still on v0.0.1 so I upgraded to v0.0.3, however now the Core2 crashes during setup(). I have narrowed this down to the createSprite() call. I have tried to debug this but I quickly get lost in the M5GFX code.

Here is a shortened version of my code:
```c++
#include <M5Unified.h>

... many other includes

#define batt_spr_wdth    190  // Battery sprite width
#define batt_spr_ht      20   // Battery sprite height

M5Canvas batt_sprite(&M5.Lcd);                        // Sprite for battery icon and percentage text

void setup(void) {
  Serial.begin(115200);
  M5.begin();
  M5.Power.setExtPower(true);  // Enable 5V output on external port

  // Create battery icon sprite
  batt_sprite.createSprite(batt_spr_wdth, batt_spr_ht);  // <== crashes here :(

  while (true); // prevent subsequent setup code from being compiled

... other code
}

```

and here is the ESP32 exception decoder output (just core 1 output). If I'm reading this correctly, the problem seems to be related to a palette colour type, as per decoder lines 6 and 5.

```
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x40091c75  PS      : 0x00060f34  A0      : 0x80093cae  A1      : 0x3ffb1e80  
A2      : 0x37735220  A3      : 0x0000abab  A4      : 0xb33fffff  A5      : 0x00000001  
A6      : 0x00060320  A7      : 0x0000cdcd  A8      : 0x400fad7c  A9      : 0x3ffb1f70  
A10     : 0x3f429568  A11     : 0x00000069  A12     : 0x3f42989c  A13     : 0x3f4295f4  
A14     : 0x00000003  A15     : 0x00060a23  SAR     : 0x0000000a  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400dbf68  LEND    : 0x400dbf72  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x40091c75:0x3ffb1e80 0x40093cab:0x3ffb1eb0 0x400942e5:0x3ffb1ed0 0x4008203d:0x3ffb1ef0 0x400d724b:0x3ffb1f10 0x400d6a39:0x3ffb1f30 0x400d1f39:0x3ffb1f50 0x400d3a35:0x3ffb1f70 0x400e1322:0x3ffb1fb0 0x40090c42:0x3ffb1fd0
  #0  0x40091c75:0x3ffb1e80 in vPortCPUAcquireMutexIntsDisabledInternal at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
      (inlined by) vPortCPUAcquireMutexIntsDisabled at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portmux_impl.h:98
      (inlined by) vTaskEnterCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:4201
  #1  0x40093cab:0x3ffb1eb0 in multi_heap_internal_lock at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:380
  #2  0x400942e5:0x3ffb1ed0 in multi_heap_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:321
  #3  0x4008203d:0x3ffb1ef0 in heap_caps_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:232
  #4  0x400d724b:0x3ffb1f10 in lgfx::v1::SpriteBuffer::reset(unsigned int, lgfx::v1::AllocationSource) at .pio/libdeps/Core2_SCD30/M5GFX/src/lgfx/v1/misc/SpriteBuffer.cpp:106
      (inlined by) lgfx::v1::SpriteBuffer::reset(unsigned int, lgfx::v1::AllocationSource) at .pio/libdeps/Core2_SCD30/M5GFX/src/lgfx/v1/misc/SpriteBuffer.cpp:131
  #5  0x400d6a39:0x3ffb1f30 in lgfx::v1::Panel_Sprite::createSprite(int, int, lgfx::v1::color_conv_t*, bool) at .pio/libdeps/Core2_SCD30/M5GFX/src/lgfx/v1/misc/colortype.hpp:585
  #6  0x400d1f39:0x3ffb1f50 in lgfx::v1::LGFX_Sprite::createSprite(int, int) at .pio/libdeps/Core2_SCD30/M5GFX/src/lgfx/v1/misc/colortype.hpp:827
  #7  0x400d3a35:0x3ffb1f70 in setup() at src/main.cpp:182
  #8  0x400e1322:0x3ffb1fb0 in loopTask(void*) at /Users/patrickfelstead/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:18
  #9  0x40090c42:0x3ffb1fd0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

```
I haven't tried the devlop branch to see if this fixes the problem yet. Please let me know if I've done something wrong, or what I can try to fix this. Thanks.

patfelst, 2021-12-03:
I just tried the develop branch code, but the problem is still there. My program includes many othe libraries. Later today I will create a new project with only M5Unified to make sure they are isolated.

lovyan03, 2021-12-03:
@patfelst 
As it turns out, removing Serial.begin(115200) will work.

In the recent update, Serial.begin has been included in the M5Unified.begin behavior.
Therefore, you do not need to do the Serial.begin.
(See HowToUse in examples).

Now, let me go one step further and explain.
Until recently, Arduino core ESP32 has a bug in Serial.end() that causes problems with running PSRAM.
And the way Serial.begin works is that if you have already begun, you call end and then begin.

This creates a situation where "calling Serial.begin twice will cause PSRAM to fail.

M5Canvas is set to use PSRAM by default.
This means that when you call createSprite, M5Canvas will access PSRAM.

Therefore, the combination of these factors will cause this phenomenon.

patfelst, 2021-12-04:
Yes that fixed it - thanks. And my apologies. I've now had a look at HowToUse example, and will implement the M5 config structure as I need set the Core2 LED, 5V output power and internal RTC etc.

closed 2021-12-04 by patfelst