m5stack/M5Unified #11: M5Core2 / M5Tough / M5StickCPlus - charging backup battery (RTC) is disabled by default

felmue, 2021-12-15:
Hi guys

is there any particular reason the backup battery (RTC) charging is disabled by default on M5Core2, M5Tough and M5StickCPlus?

I only noticed this since the backup battery in my M5Tough was flat when the device arrived and the RTC time restarted at 00:00:00 every time I powered M5Tough via USB and it did not improve even after letting it run with USB plugged in. Then looking through the code I noticed this function `Axp192.setBACKUP();` but it isn't called from anywhere.

Thanks
Felix

lovyan03, 2021-12-15:
@felmue 
hello, thanks for your report .

The configuration of AXP192 at startup is done in the following line of Power_Class.cpp.

https://github.com/m5stack/M5Unified/blob/0.0.3/src/utility/Power_Class.cpp#L191

In order to reduce the program size, the number and value of each register is stored in an array instead of a normal function call.

When I tested, register 0x35 is set to the value 0xA2.
This is the same value that is initially set in the setBACKUP function and the M5Core2 and M5StickC libraries.

The following code can display all the registers of the AXP192, and just to be sure, make sure that the value of register 0x35 is 0xA2.

If you have tried this and found that the value of register 0x35 is 0xA2, then there is probably another reason why the RTC backup is not working.

```
#include <M5Unified.h>

void setup(void)
{
  M5.begin();
  if (M5.Display.width() < M5.Display.height())
  { /// Landscape mode.
    M5.Display.setRotation(M5.Display.getRotation() ^ 1);
  }
  M5.Display.setTextColor(TFT_GREEN, TFT_BLACK);
  for (int x = 0; x < 16; ++x) {
    M5.Display.setCursor((x+1) * M5.Display.width() / 17, 0);
    M5.Display.printf("_%x", x);
  }
  for (int y = 0; y < 10; ++y) {
    M5.Display.setCursor(0, (y+1) * 10);
    M5.Display.printf("%x_", y);
  }
  M5.Display.setTextColor(TFT_WHITE, TFT_BLACK);
}

void loop(void)
{
  uint8_t buf[128];
  M5.In_I2C.start(0x34, 0, 400000);
  M5.In_I2C.write(0);
  M5.In_I2C.restart(0x34, 1, 400000);
  M5.In_I2C.read(buf, 128);
  M5.In_I2C.stop();

  M5.Display.startWrite();
  for (int y = 0; y < 10; ++y) {
    for (int x = 0; x < 16; ++x) {
      int i = (y << 4) + x;
      M5.Display.setCursor((x+1) * M5.Display.width() / 17, (y+1) * 10);
      M5.Display.printf("%02x", buf[i]);
    }
  }
  M5.Display.endWrite();

  static constexpr const char* const wd[7] = {"Sun","Mon","Tue","Wed","Thr","Fri","Sat"};

  if (!M5.Rtc.isEnabled())
  {
    M5.Rtc.begin();
  }
  else
  {
    m5::rtc_datetime_t dt;
    if (M5.Rtc.getDateTime(&dt))
    {
      char buf[32];
      snprintf( buf, 30, "%04d/%02d/%02d(%s)"
              , dt.date.year
              , dt.date.month
              , dt.date.date
              , wd[dt.date.weekDay]
              );
      M5.Display.drawString(buf, 0, M5.Display.height() - 20);
      snprintf( buf, 30, "%02d:%02d:%02d"
              , dt.time.hours
              , dt.time.minutes
              , dt.time.seconds
              );
      M5.Display.drawString(buf, 0, M5.Display.height() - 10);
    }
  }
  m5gfx::delay(50);
}
```

felmue, 2021-12-16:
Hello @lovyan03 

first of all thank you very much for the detailed information and explanation. I appreciate that.

I compiled an ran the above code you provided and yes, register 0x35 is properly set to 0xA2.

However when I re-download the stock factory image my M5Tough came with originally and then only check register 0x35 (w/o writing it first) it has the default value of 0x22. (Which would match my original observation, that the RTC backup battery wasn't being charged and its voltage stayed at 0.7 V.)

I must assume that the stock factory image, loaded onto the M5Tough in the factory, did not yet contain the code to set register 0x35 to 0xA2. Or in other words the stock factory image was built from an earlier version of the M5Unified code.

Thanks
Felix

lovyan03, 2021-12-16:
@felmue 
Thanks for the information !
It is true that the development of the shipping firmware for the M5Tough and the M5Unified were done at the same time, so there must have been some imperfections.
I would like to proceed with the preparation to update the [factory firmware](https://github.com/m5stack/M5Tools/issues/1).

felmue, 2021-12-16:
Hello @lovyan03 

thank you for taking care of that. I am sure your customers will appreciate the RTC backup battery being charged in their new M5Tough.

Thanks
Felix

lovyan03, 2021-12-24:
@felmue 
M5Tools v1.0.1 is now available for M5Burner.
The fixes have been reflected, so you can try it out if you like.

felmue, 2021-12-25:
Hello @lovyan03 

thank you for letting me know. I appreciate it.

Yes, I can confirm that M5Tools 1.0.1 correctly set register 0x35 to 0xA2. I've also checked the RTC backup battery and I can see the voltage rising, e.g. it is being charged.

Thanks
Felix

closed 2022-04-04 by lovyan03