m5stack/M5Unified #13: How to reconfigure pins on second I2C channel

patfelst, 2021-12-23:
Hi there, for reasons I won't go into here, I'm trying to initialise the second I2C port as follows:

SDA = GPIO26
SCL = GPIO36

The first thing I do is:
```C++
void setup(void) {
  auto cfg = M5.config();
  cfg.serial_baudrate = 115200;  // default=115200. if "Serial" is not needed, set it to 0.
  cfg.clear_display = true;      // default=true. clear the screen when begin.
  cfg.output_power = true;       // default=true. use external port 5V output.
  cfg.internal_imu = true;      // default=true. use internal IMU.
  cfg.internal_rtc = true;       // default=true. use internal RTC.
  cfg.external_imu = false;      // default=false. use Unit Accel & Gyro.
  cfg.external_rtc = false;      // default=false. use Unit RTC.
  cfg.led_brightness = 0;        // default= 0. Green LED brightness (0=off / 255=max) (※ not NeoPixel)

  M5.begin(cfg);
...
```
and then a short while later, I call

`    begin_ok = Wire.begin(26, 36);
`

However my I2C sensor on pins 26, 26 is not being found. I think this should be possibe, but the M5Unified begin code is a little hard for me to follow. It appears to me that you use I2C channel 1 (I2C_NUM_1) pins 21 and 22 for the PMU and internal RTC etc. So I2C channel 0 should be available for me to use given I'm not using external RTC or external IMU.

Is there an easy way for me to configure the unused I2C port with pins 26 and 36?

Thanks in advance!

lovyan03, 2021-12-23:
Hello. @patfelst .
As it turns out, the GPIOs that can be used for both Read and Write are from 0 to 33.
GPIO34 and above are read-only and cannot be used for I2C.

If you want to use GPIO26 and 36, I guess you want to use the HAT port of M5StickC or CoreInk.
You should be able to use GPIO0, 25, 26 in any combination.

patfelst, 2021-12-23:
Oh that's embarrasing...I did check the ESP32 datasheet, but missed that input only on GPIO36. Assuming I use other pins from 0 to 33, would the configuration of using Wire.begin(pinA, pinB) work ok with M5Unified?
Thank you.

closed 2021-12-23 by patfelst