m5stack/M5Unified #19: GoPlus2 problem

bitscollector, 2022-03-02:
Hi, I tried the library with two samples and and when I run the programs in a Core2 with a GoPlus2 attached, the servo connected to SERVO1 out start to run.

The examples I tried are the bluetooth speaker and the "HowToUse.ino" demo.

lovyan03, 2022-03-02:
@bitscollector 
Sorry, Core2 and GoPlus2 cannot be combined.
Since both the touch panel controller and GoPlus2 have I2C address 0x38, we cannot guarantee the operation.

https://docs.m5stack.com/en/module/goplus2?id=goplus2
[image]

bitscollector, 2022-03-03:
I want to thank to you for writing this library. When I bought the module, the compatibility advice wasn't in the web. Anyway, I'm not an advanced user nor a programmer and I have a question: Do you think I could deactivate the panel touch driver avoiding conflicts? I ask because I used the module with UiFlow with no problems other than the touch panel don't working.

Thanks in advance and thanks again for your work and for reply in english. I'm not english speaker but at least can understand a little.

lovyan03, 2022-03-03:
@bitscollector 
Probably possible.
The touch panel can be deactivated by writing 0x03 to touch panel register 0xA5.
However, since M5GFX currently uses the presence or absence of touch panel response as a factor in determining the model between Core2 and M5Tough, we have found that the touch panel is misidentified as M5Tough after resetting.

Please give me some time to make adjustments.

bitscollector, 2022-03-03:
I'm sorry I have not the knowledge to help :/ but if you need I test something with the GoPlus2 just tell me.

lovyan03, 2022-03-04:
@bitscollector 

Thank you for your patience.

M5Unified and M5GFX updated on develop branch .
This update has not been officially released yet, so please get it from the develop branch and try it out.

I believe the touch panel can be deactivated by executing the following setup code.

The content of the loop is an I2C scanner. The address of a working I2C device is highlighted.
Run this code with GoPlus2 removed, and if the 0x38 response is gone, you have succeeded.

If it works, the next step is to connect the GoPlus2 and turn it back on.
There is a 0x38 response from GoPlus2, but since the touch panel is disabled, it probably will not affect the servo's operation.

```
#include <M5Unified.h>

void setup(void) {
  M5.begin();

  // TouchPanel(0x38) Register(0xA5) = hibernate mode(0x03)  (I2C 40000Hz)
  M5.In_I2C.writeRegister8(0x38, 0xA5, 0x03, 400000);

  // dispose touch class;
  M5.Touch.end();

  // Wait 30 milliseconds for touch screen to go to sleep
  delay(30);

  // Need to update here.
  M5.update();
}

void loop(void) {
  /// I2C scanner

  bool scan[128];
  M5.In_I2C.scanID(scan);
  M5.Display.startWrite();
  for (int i = 8; i < 0x78; ++i) {
    M5.Display.setTextColor(TFT_DARKGRAY, TFT_BLACK);
    if (scan[i])   {
      M5.Display.setTextColor(TFT_BLACK, TFT_YELLOW);
    }
    M5.Display.setCursor( ((i & 15) * M5.Display.width()) >> 4, (i >> 4) * 10);
    M5.Display.printf("%02x", i);
  }
  M5.Display.endWrite();
  delay(100);
}
```

bitscollector, 2022-03-04:
Hi again.
When I run the code, using the developer branch of the library with GoPlus2 removed. Only 34 and 51 are highlighted. When attach the GoPlus2, the 38 and 68 are highlighted too and then the motor start to run.

lovyan03, 2022-03-05:
@bitscollector 
Hmmm...I need to see what triggers GoPlus2 to work.
What happens if comment out the following line of code from earlier?

```
// M5.In_I2C.writeRegister8(0x38, 0xA5, 0x03, 400000);
```

bitscollector, 2022-03-05:
I commented that line and tried again with GoPlus2 removed. The first time the 38 was highlighted. Tried again with the line uncommented and then again commenting the line and this time the screen gone black and Core2 started to vibrate. I didn't even know the Core2 have a vibration motor.

Then tried again with the line commented, the 38 was highlighted and, if attach the GoPlus2, the servo start to run.

Thank again for your work and don't worry to take the time you need.

lovyan03, 2022-03-05:
@bitscollector 
Oh...the screen going black out and the vibration motor running means that the M5GFX's auto-detection function misidentified it as an M5Tough. That means you are still not using the develop branch that I updated for you.

> M5Unified and M5GFX updated on develop branch .
> This update has not been officially released yet, so please get it from the develop branch and try it out.

M5Unified develop download:
https://github.com/m5stack/M5Unified/archive/refs/heads/develop.zip

M5GFX develop download:
https://github.com/m5stack/M5GFX/archive/refs/heads/develop.zip

Try again with these two libraries.

lovyan03, 2022-04-03:
@bitscollector 
How are you doing since then?
The library has officially released an upgrade.
If you would like, please update the library and try again.

closed 2022-04-07 by lovyan03