m5stack/M5Unified #22: Web Radio sample compile error

kawashimaken, 2022-04-03:
いつもお世話になっています。
WebRadioのサンプルコードですが
コンパイルすると下記のエラーになります。
ESP8266Audioも他のライブラリ、ボードライブラリもArduinoIDE全て最新にしています。
`
In file included from /Users/demo/Documents/Arduino/libraries/ESP8266Audio/src/AudioGeneratorMIDI.cpp:65:
/Users/demo/Documents/Arduino/libraries/ESP8266Audio/src/libtinysoundfont/tsf.h: In function 'void tsf_channel_midi_control(tsf*, int, int, int)':
/Users/demo/Documents/Arduino/libraries/ESP8266Audio/src/libtinysoundfont/tsf.h:2100:1: error: insn does not satisfy its constraints:
 }
 ^
(insn 858 343 344 51 (set (reg:SF 19 f0 [407])
        (mem/u/c:SF (symbol_ref/u:SI ("*.LC248") [flags 0x2]) [0  S4 A32])) "/Users/demo/Documents/Arduino/libraries/ESP8266Audio/src/libtinysoundfont/tsf.h":2053 47 {movsf_internal}
     (nil))
during RTL pass: postreload
`

lovyan03, 2022-04-04:
@kawashimaken 
報告ありがとうございます。

ESP8266Audio の AudioGeneratorMIDI は ArduinoESP32のv2.0.1 , v2.0.2 では
コンパイラのバグによってビルドに失敗するとの報告があるようです。

https://github.com/earlephilhower/ESP8266Audio/pull/502

上記内容によると、問題のあるバージョンにおいては AudioGeneratorMIDIを
ビルド対象から外す対策が取られたようです。
なおこの対策内容は、現時点では まだリリースバージョンに反映されていないようです。

従いまして 本件の対策としては、ESP8266Audioをgithubのマスターから取得するか、
またはArduinoESP32の v1.0.6を使用すること、となります。
よろしくお願いいたします。

kawashimaken, 2022-04-04:
@lovyan03  早速お返事いただきありがとうございます！
ArduinoESP32の v1.0.6にしたら、問題なくコンパイルが出来ました！

ただ、最後PATHにpythonが見つかれみませんというメッセージが出ており、これは私の環境の問題（MacOS Monterey）かと思いますので、解決法を別途調べます。この件は一旦クローズさせていただきます！ありがとうございます！

closed 2022-04-04 by kawashimaken