m5stack/M5Unified #24: Vibration Motor

robcloot, 2022-04-17:
How do I turn on the Vibration Motor (PWM:AXP_LDO3)?

lovyan03, 2022-04-17:
@robcloot 
The following functions are available

```
  M5.Power.Axp192.setLDO3(3300);
```

The argument is in millivolts. it can be set in increments of 100, with a maximum of 3300.
If there is a possibility of running on more than one model, it is recommended to run the program after model identification.

```
  if (M5.getBoard() == m5::board_t::board_M5StackCore2)
  {
    M5.Power.Axp192.setLDO3(3300);
    delay(200);
    M5.Power.Axp192.setLDO3(0);
  }
```

robcloot, 2022-04-18:
Thank you!  That works great!
1. How do I get a list of all the M5Unified Functions?
2. How do I send you a donation for creating the M5Unified library?

patfelst, 2022-04-18:
I look at the examples, and also delve into the library code headers. There are some API docs here, but I'm not sure if they're in sync with the M5Unified library https://docs.m5stack.com/en/api/core/system

lovyan03, 2022-04-18:
@robcloot 
I apologize that the API documentation has not been prepared.
Because M5Unified is a latecomer, almost all API information available on the web is from conventional (M5Stack.h, M5Core2.h, etc.) libraries.
We recognize the urgent need for documentation.
At this time, the only way to understand it is to read the header files and run the sample code.

Your offer of a donation would be greatly appreciated, but we would prefer that you support us by purchasing M5Stack products.
Since this is the official M5Stack repository, I respectfully decline to personally accept donations.

robcloot, 2022-04-18:
Thank you again. If you continue to creat code and support the software, then M5Stack products will be selling off the shelf. You have a great product!  Thank you for answering my questions.

closed 2023-02-08 by lovyan03

Niclassslua, 2024-08-09:
Quick question, I am using a Core 2 and I'm trying to use the vibration motor. With M5Core2.h it works. When trying the M5Unified though it doesn't work... I am using this code:

if (M5.getBoard() == m5::board_t::board_M5StackCore2)
  {
    M5.Power.Axp192.setLDO3(3300);
    delay(200);
    M5.Power.Axp192.setLDO3(0);
  }