m5stack/M5Unified #25: M5GO2 (for CORE2, AWS)

armel, 2022-04-19:
Hi,

Thank you for your incredible work. 
The leds from M5GO2 module seem to don't work (GPIO 25) with M5Stack CORE2 or AWS. 
It works like a charm with M5GO module on BASIC (GPIO 15). 
Do you confirm ?

Armel.

lovyan03, 2022-04-19:
@armel 
Sorry, I don't understand the problem as I don't know the code you ran.
Can you give me some more details?

Perhaps you want to control RGB LEDs, but M5Unified does not yet implement RGB LED functionality.
(I plan to add that functionality in the near future)

armel, 2022-04-19:
Excuse me, I think I understood my mistake. I'm sorry to have disturbed you. You can close this issue. And thank you again for your great work 👍🏼 

Armel.

closed 2022-04-19 by armel