m5stack/M5Unified #27: Cannot compile RTC.ino example

patfelst, 2022-05-01:
Hi there, thanks for your great library. My question is related to PR #20 

I cannot get the basic [RTC.ino](https://github.com/m5stack/M5Unified/blob/master/examples/Basic/Rtc/Rtc.ino) example to compile when I uncomment these lines to sync Core2 RTC with NTP
https://github.com/m5stack/M5Unified/blob/42717f09c24e5af3f8edeef097c2755fdc36b571/examples/Basic/Rtc/Rtc.ino#L42-L66

It is failing on line 55. I'm unsure how this can compile when it is an IDF framework function, not an Arduino framework function?
[  while (sntp_get_sync_status() != SNTP_SYNC_STATUS_COMPLETED)](https://github.com/m5stack/M5Unified/blob/42717f09c24e5af3f8edeef097c2755fdc36b571/examples/Basic/Rtc/Rtc.ino#L55)

I acknowledge this is not a M5 Unified issue, but any help would be appreciated because I get inconsistent results using a 1sec delay after calling sync with NTP. Sometimes it works, other times it syncs to an incorrect time.

Thankyou.

lovyan03, 2022-05-01:
@patfelst 
Thanks for reporting this Issue.
It seems that there was a missing header file that should be included.
Please try adding the following code to the top of the sample code.

```
#include <esp_sntp.h>
```

patfelst, 2022-05-01:
Yes thanks, that fixed the problem! And it works reliably now, good NTP sync every time!

In case anyone else is reading this, for my home location in Adelaide, this is the function call that's working for me:
`  configTzTime("ACST-9:30ACDT,M10.1.0,M4.1.0/3", "pool.ntp.org");`

closed 2022-05-01 by patfelst