m5stack/M5Unified #28: The TF card based on arduinoM5core2 does not work properly？

yihua-wang, 2022-05-10:
conditions：
1.use M5Unified lib；
2.use platfrom io IDE;
3.use Arduino Framwork base Espressif 32 Verision 4.2.0 lasted.

lovyan03, 2022-05-10:
Hello @yihua-wang .
M5Unified itself does not have the ability to handle SD cards, so I am not familiar with the situation.
Please present a program that can reproduce the Issue you encountered.

tanakamasayuki, 2022-05-10:
Hi.

https://gist.github.com/tanakamasayuki/9f814c02928b6c39a2644734ba670ebe

When tested in the past, it didn't work with 4.2.0
I think it worked with 4.1.0

I think it's a bug in Arduino 2.0.2.
2.0.3 is in preparation.

The recommendation is 1.0.6.

platform = espressif32       ;ESP-IDF(4.4)  , Arduino(2.0.2)
platform = espressif32@4.2.0 ;ESP-IDF(4.4)  , Arduino(2.0.2)
platform = espressif32@4.1.0 ;ESP-IDF(4.4)  , Arduino(2.0.1)
platform = espressif32@4.0.0 ;ESP-IDF(4.4)  , Arduino(2.0.0)
platform = espressif32@3.5.0 ;ESP-IDF(3.3.5), Arduino(1.0.6)

closed 2023-02-08 by lovyan03