m5stack/M5Unified #33: channel default volume is unusual

belm0, 2022-07-17:
I was surprised that the channel default volume is 64.  In audio designs, usually channel volume is attenuation only.  So I'd expect the default to be the maximum (255).

Otherwise, the master volume doesn't really mean the "maximum volume", and some channels may have a volume equivalent to a master volume of 1024...

https://github.com/m5stack/M5Unified/blob/42717f09c24e5af3f8edeef097c2755fdc36b571/src/utility/Speaker_Class.hpp#L234

lovyan03, 2022-07-17:
@belm0 
Thank you for your important input.
I apologize for any inconvenience caused by my ignorance of standard audio designs thinking.
I will review the volume settings in the next version and set the default channel volume to 255.

By the way, the volume value is calculated internally as a square ( `volume ^ 2` ) . In other words, the output varies quadratically.
Both master volume and channel volume.
Do you have any opinions or generalizations about this process?

belm0, 2022-07-17:
> By the way, the volume value is calculated internally as a square ( volume ^ 2 )

I was going to say something about this requiring a 64-bit intermediate value for each sample unless some optimizations are done, but then I realized that the code is using a float buffer.  I'm a little surprised because float is not very fast on ESP32, so I guess this is a luxury sound library 😉

lovyan03, 2022-07-17:
@belm0 
Oh.. yes, I was reluctant to use floats, but I couldn't think of a better way.
In fact, I've been racking my brains because I later realized that the float operation is even slower in ESP32C3…😨

belm0, 2022-07-18:
I have no doubt that multi-channel, 16-bit sound can be handled well within fixed-point, 32-bit operations.  (The PlayStation or early consoles and computers had no floating point hardware, etc.)

To start, 32-bit dynamic range is really not needed for volume.