m5stack/M5Unified #34: Trouble using SmartConfig

kotobuki, 2022-07-28:
Hi. Thank you very much for releasing such a useful library. 🙇

I have been trying to migrate from the M5Stack library to the M5Unified library and found a problem regarding SmartConfig. The following example works fine when `M5Stack.h` is selected. However, I can't configure Wi-Fi when `M5Unified.h` is selected. I confirmed with an M5Stack FIRE. 

```C++
#include <WiFi.h>

#if 1
#include <M5Unified.h>
#else
#include <M5Stack.h>
#endif

void setup(void) {
  M5.begin();

  WiFi.mode(WIFI_AP_STA);
  WiFi.beginSmartConfig();

  M5.Lcd.println("Waiting for SmartConfig");
  while (!WiFi.smartConfigDone()) {
    delay(500);
    M5.Lcd.print(".");
  }

  M5.Lcd.println();

  M5.Lcd.println("Waiting for Wi-Fi");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    M5.Lcd.print(".");
  }

  M5.Lcd.println();

  M5.Lcd.print("IP Address: ");
  M5.Lcd.println(WiFi.localIP());
}

void loop(void) {
  vTaskDelay(1);
  M5.update();
}
```

I'm sorry if this is a false alarm, but I want to report to make sure. Thank you very much.

kotobuki, 2022-07-28:
I'm sorry, I forgot to report versions.

- Board: M5Stack v2.0.3
- Library: M5Stack v0.4.1, M5Unified v0.0.7

lovyan03, 2022-07-28:
@kotobuki 
Thank you for your report.
I have tried using the source code you provided and it works without any problems.
However, regardless of M5Unified, SmartConfig itself sometimes does not reflect the settings immediately.

Currently, I think there is not enough information to determine if this is caused by M5Unified.
I'm sorry, but I can't think of any ideas to address this right now.

kotobuki, 2022-07-28:
@lovyan03
Thank you very much for your prompt reply. 🙇

To make sure, I removed the library folder and installed the M5Unified library v0.0.7 and the M5GFX as requested, then installed the M5Stack library v0.4.1 and required libraries. Still, I can reproduce this issue (i.e. SmartConfig with EspTouch works fine when M5Stack.h enabled but fails when M5Unified.h enabled). For your reference, I want to share a part of a compilation log.

```
Creating esp32 image...
Merged 25 ELF sections
Successfully created esp32 image.
python3 /Users/{my_user_name}/Library/Arduino15/packages/m5stack/hardware/esp32/2.0.3/tools/gen_esp32part.py -q /var/folders/3h/svk912wd4g72lp2ks9mtx99c0000gn/T/arduino_build_582839/partitions.csv /var/folders/3h/svk912wd4g72lp2ks9mtx99c0000gn/T/arduino_build_582839/sketch_jul28a.ino.partitions.bin
Multiple libraries were found for "WiFi.h"
 Used: /Users/{my_user_name}/Library/Arduino15/packages/m5stack/hardware/esp32/2.0.3/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
Using library WiFi at version 2.0.0 in folder: /Users/{my_user_name}/Library/Arduino15/packages/m5stack/hardware/esp32/2.0.3/libraries/WiFi 
Using library M5Unified at version 0.0.7 in folder: /Users/{my_user_name}/Documents/Arduino/libraries/M5Unified 
Using library M5GFX at version 0.0.20 in folder: /Users/{my_user_name}/Documents/Arduino/libraries/M5GFX 
Using library FS at version 2.0.0 in folder: /Users/{my_user_name}/Library/Arduino15/packages/m5stack/hardware/esp32/2.0.3/libraries/FS 
Using library SPI at version 2.0.0 in folder: /Users/{my_user_name}/Library/Arduino15/packages/m5stack/hardware/esp32/2.0.3/libraries/SPI 
Using library Wire at version 2.0.0 in folder: /Users/{my_user_name}/Library/Arduino15/packages/m5stack/hardware/esp32/2.0.3/libraries/Wire 
/Users/{my_user_name}/Library/Arduino15/packages/m5stack/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/xtensa-esp32-elf-size -A /var/folders/3h/svk912wd4g72lp2ks9mtx99c0000gn/T/arduino_build_582839/sketch_jul28a.ino.elf
Sketch uses 813125 bytes (62%) of program storage space. Maximum is 1310720 bytes.
Global variables use 41312 bytes (12%) of dynamic memory, leaving 286368 bytes for local variables. Maximum is 327680 bytes.
```

As you can identify from the log, I have been using ESP32 Arduino Core v2.0.3 as a part of the M5Stack board v2.0.3. To make sure, I installed v2.0.4, but the result was the same. 🤔

I'll try creating another development environment on Windows and report here again.

kotobuki, 2022-07-29:
I just created another development environment (the versions of the board and libraries are the same) on Windows 10 and confirmed that this issue is reproducible. If there is anything I can try, please let me know. Thank you very much in advance.

lovyan03, 2022-07-29:
@kotobuki 
I’ｍ sorry... I have no idea what is going on in your environment...

I have run the library with the version you provided, but I cannot reproduce the problem as follows.
I need you to ERASE the ESP32 flash once and see if the situation improves.

Do you have any other M5Stack products that you can test?
If so, can you confirm if the problem occurs with other units?

[image]

[image]

kotobuki, 2022-07-29:
@lovyan03 Thank you very much for taking the time to do testing on your side. 🙇

I tried again with the M5Stack FIRE after erasing the flash, and the result was the same. However, I tried with an M5Stack Core2 and successfully connected with SmartConfig.

I'll try with another device and report here once I find the cause of this issue.

lovyan03, 2022-07-29:
@kotobuki 
I have been informed that when the M5GO bottom is installed, the state of GPIO15 affects the WiFi behavior.
Please add the following line and test with FIRE before running the smart config.

```
pinMode(15, OUTPUT_OPEN_DRAIN);
```

If this resolves the issue, we will implement a fix to include this in the M5Unified initialization process.

kotobuki, 2022-07-29:
@lovyan03
Thank you very much! I modified the test code and successfully configured Wi-Fi for the M5Stack FIRE several times with EspTouch. It seems that the change made a difference. 😉 

```C++
#include <WiFi.h>

#if 1
#include <M5Unified.h>
#else
#include <M5Stack.h>
#endif

void setup(void) {
  M5.begin();
  pinMode(15, OUTPUT_OPEN_DRAIN);

  WiFi.mode(WIFI_AP_STA);
  WiFi.beginSmartConfig();
```

lovyan03, 2022-07-29:
@kotobuki 
Thanks for your help in validating this!
I will add the relevant code in the init function of M5Unified shortly.
I believe this will be reflected in the next version release, so until then, please take care of this on the user code side.

lovyan03, 2023-02-08:
This Issue has been resolved in the current version, so I am closing this Issue. Thank you very much.

closed 2023-02-08 by lovyan03