m5stack/M5Unified #36: Panic on M5Tough

maxammann, 2022-08-05:
The M5Tough is crashing with the following panic during the Arduino setup() function:

```
I (0) cpu_start: Starting scheduler on APP CPU.
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x00000040  PS      : 0x00060030  A0      : 0x800d5fa5  A1      : 0x3ffb9eb0  
A2      : 0x3ffb2cf4  A3      : 0x00000040  A4      : 0x000000a4  A5      : 0x00000000  
A6      : 0x00010101  A7      : 0x00060823  A8      : 0x800d6aa7  A9      : 0x3ffb9ea0  
A10     : 0x3ffb2cf4  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffb32da  
A14     : 0x00000000  A15     : 0x01000101  SAR     : 0x00000000  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000040  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace:0x0000003d:0x3ffb9eb00x400d5fa2:0x3ffb9f50 0x400894e5:0x3ffb9f70 
0x400d5fa2: loopTask(void*) at /home/max/projects/hardware/esp32-temp/tough/build/../components/arduino-esp32/cores/esp32/main.cpp:42

0x400894e5: vPortTaskWrapper at /home/max/esp/esp-idf-v4.4/components/freertos/port/xtensa/port.c:131
```

I'm trying to run the `HowToStart` example.

I was able to draw to the LCD using the M5Tough library.

## Used Software

M5Unified: Latest as of today
ESP-IDF: 4.4
arduino-esp32: Latest as of today (master branch)

maxammann, 2022-08-05:
Right after posting this issue I noticed that I would try to run without ardunio-esp32 as M5Unified supports that. And it seems to work!

So now the only question would be: Why is arduinio-esp32 interfering here? What is breaking?

lovyan03, 2022-08-06:
@maxammann 
I'm sorry, but since I don't know the source code you ran, I have no idea what caused the problem…

maxammann, 2022-08-06:
I created an example to reproduce: https://github.com/ammannbros/esp32-temperature/tree/repro-panic

Build & flash using:

```
idf.py build
idf.py flash
```

lovyan03, 2022-08-06:
Could you please create a code that can reproduce the phenomenon in a short number of lines, using the ArduinoESP32, which is now officially released?

maxammann, 2022-08-06:
As I wrote in the issue, the problem exists together with the current arduino-esp32.
The repository I posted should allow you to reproduce the issue easily. Just make sure to clone submodules recursively.

maxammann, 2022-08-06:
The esp crashes as soon as `M5.begin()` is executed.

lovyan03, 2022-08-06:
I have looked at the structure of your project and it deviates from the way M5Unified is intended to be used.
It is not intended to be built with ESP-IDF with ArduinoESP32 as a sub-module.
Therefore, I cannot devote my time to debugging this project.

please use ArduinoIDE or use PlatformIO with `framework = arduino` specified.

maxammann, 2022-08-06:
Oke I though that ESP-IDF as such is supported. If that is not the case I would propose to update the README.md. I opened a PR.

Possibly the product page of the M5Touch should also be updated: https://shop.m5stack.com/products/m5stack-tough-esp32-iot-development-board-kit

maxammann, 2022-08-06:
Nonetheless, the setup I went with should work. Therefore, there definitely is a bug somewhere in M5Unified.
Sooner or later it will probably come up again.

closed 2022-08-06 by maxammann

lovyan03, 2022-08-07:
The develop branch of M5Unified and M5GFX has been updated.
This modification adds the following comment line in CMakeLists.txt

```
# list(APPEND COMPONENT_REQUIRES arduino-esp32)
```

If you want to use it with arduino-esp32 component, please uncomment this comment and enable it.
This must be done for both M5GFX and M5Unified.