m5stack/M5Unified #39: SD Card doesn't get mounted from the M5Unified Library [Core2]

FateEscape, 2022-08-29:
Hi there

When changing from the Core2 Library to the M5Unified library, since it allows to create sin-waves via speaker, the SD-Card isn't recognized, probably by <SPIFFS.h> or <SD.h>.

#include <M5Unified.h>
#include <SPIFFS.h>
#include <SD.h>

I thought it might be a problem with the format of the SD-Card being on FAT32, but changing it to NTFS didn't solve the problem.

Monitoring Error Code: [ 10910][E][vfs_api.cpp:24] open(): File system is not mounted

lovyan03, 2022-08-29:
Sorry, M5Unified does not yet implement operations on SD cards.
Therefore, you need to implement `SD.begin` in your user code.

FateEscape, 2022-08-29:
After implementing the SD.begin() i get another Error message: 

`[290459][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00`
`[290459][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work`

The SD-Card i'm using is a SDHC, might there be a problem with the SD-Card itself?
Since it worked with Core2 Library i formated it back to FAT32

My Code:
```
#include <M5Unified.h>
#include <driver/i2s.h>
#include <driver/i2c.h> 
#include <SPIFFS.h>
#include <SD.h>
File myFile;

void setup() {

  M5.begin();
  M5.Speaker.begin();
  SD.begin();

File myFile;
  String ConfigLine;
  myFile = SD.open("/testing/device.txt");
  if (myFile) {
    while (myFile.available()) {
      ConfigLine = myFile.readStringUntil('\r');  //read complete line
      Serial.println(ConfigLine);
      MachineID = ConfigLine;

    }
    myFile.close();
  }

  Serial.println("ReadDevice Finished");
    Serial.println("MachineID = " + MachineID);

}

void loop()
{
  M5.update();
}
```

Is there something wrong with the way of the implementation? 
I'm quite new to using Core2 and Arduino, help is very much appreciated

lovyan03, 2022-08-29:
How about the following ?
```
 while (!SD.begin(GPIO_NUM_4, SPI, 25000000))
 {
   delay(1);
 }
```

FateEscape, 2022-08-29:
Now it works, thank you very much @lovyan03 😄

closed 2022-08-29 by FateEscape