m5stack/M5Unified #40: SD-Card (GPIO_NUM_4) interferes with DAC Output

FateEscape, 2022-08-31:
Hi there

When starting the implementation of the SD-Card with `SD.begin(GPIO_NUM_4, SPI, 25000000)`, the DAC Output of G25 and G26 
aren't giving any signals out.

I tested it with an Oscilloscope which is able to detect Frequency's.
The DAC Output normally works when the SD-Card isn't implemented, so it seems that GPIO_NUM_4 interferes with DAC in some way.

Is there a way to set the GPIO_NUM_4 back to it's origin settings after the Data has been downloaded and loaded from the SD-Card, or is there another solution to it? 

The settings of Speaker_Class Config is:

```
struct speaker_config_t
  {
    /// i2s_data_out (for spk)
    int pin_data_out = I2S_PIN_NO_CHANGE;

    /// i2s_bck
    int pin_bck = I2S_PIN_NO_CHANGE;

    /// i2s_ws (lrck)
    int pin_ws = I2S_PIN_NO_CHANGE;

    /// output sampling rate (Hz)
    uint32_t sample_rate = 254000;

    /// use stereo output
    bool stereo = false;

    /// use single gpio buzzer, ( need only pin_data_out )
    bool buzzer = false;

    /// use DAC speaker, ( need only pin_data_out ) ( only GPIO_NUM_25 or GPIO_NUM_26 )
    bool use_dac = true;

    /// Zero level reference value when using DAC ( 0=Dynamic change )
    uint8_t dac_zero_level = 0;

    /// multiplier for output value
    uint8_t magnification = 128;

    /// for I2S dma_buf_len
    size_t dma_buf_len = 128;

    /// for I2S dma_buf_count
    size_t dma_buf_count = 8;

    /// background task priority
    UBaseType_t task_priority = 2;

    /// background task pinned core
    BaseType_t task_pinned_core = -1;

    /// I2S port
    i2s_port_t i2s_port = i2s_port_t::I2S_NUM_0;
  };
```

lovyan03, 2022-08-31:
Please set this up.
```
config.pin_data_out = 25;
```

lovyan03, 2022-08-31:
and, perhaps `magnification = 128` is too big...

lovyan03, 2022-08-31:
@FateEscape 
... I have a few concerns that I would like to confirm.
Are you using Core2?
And do you want to use a DAC to output to external speakers instead of the built-in speakers?

FateEscape, 2022-08-31:
@lovyan03 
Sorry for taking a bit long to respond back, i was testing and trying to find where the issue was.

> and, perhaps `magnification = 128` is too big...

With that i wanted to see how the magnification influences the sin_waves when using the Oscilloscope

> @FateEscape ... I have a few concerns that I would like to confirm. Are you using Core2? And do you want to use a DAC to output to external speakers instead of the built-in speakers?

I use the Core2 and use DAC to detect Sin-Waves in a Oscilloscope, that might count as an external speaker

**It actually works now!**
After testing and searching for the issue i found out that a code i used to play a mp3 file was interfering with DAC and probably also the build-in speakers.

The issue was:

```
file = new AudioFileSourceSD("/sounds/DE/success.mp3");
    id3 = new AudioFileSourceID3(file);
    out = new AudioOutputI2S(0, 0); // Output to builtInDAC
    out->SetPinout(12, 0, 2);
    out->SetOutputModeMono(true);
    out->SetGain((float)OUTPUT_GAIN / 100.0);
    mp3 = new AudioGeneratorMP3();
    mp3->begin(id3, out);
```

Deleting that solved it. Question now would be how to play mp3 files and still use DAC afterwards, but since that isn't mandatory in my case it's fine how it is now.

Setting the Data pin to 25 manually wasn't actually mandatory, so that issue went on my side, sorry for that.

https://user-images.githubusercontent.com/69121451/187642439-3099af76-1dd7-4c3e-881d-4b5eaa71e5e6.mp4

The Data from the SD-Card is loaded and the Sin-Waves are created and outputted perfectly, the quality of the waves are awesome! 

Code to create those waves: 

  ```
M5.Speaker.setVolume(100);
  M5.Speaker.tone(1000, 0, 0, false, sin_wav, sizeof(sin_wav));
  M5.Speaker.tone(1001, 0, 1, false, sin_wav, sizeof(sin_wav));
  M5.Speaker.tone(1002, 0, 2, false, sin_wav, sizeof(sin_wav));
```

What would be the limit of the sample_rate? It's currently on 254'000.

lovyan03, 2022-08-31:
@FateEscape 
I do not know what the performance limits of the ESP32's DAC are, but it can operate up to at least 1 MHz.

BTW, Perhaps you are editing the contents of the library directly?
Instead of rewriting the library, you can do the configuration on the user code side.
Try the following

```
#include <M5Unified.h>

// pseudo sin wave data
static constexpr uint8_t sin_wav[] = { 177, 245, 245, 177, 79, 11, 11, 79 }; 

void setup(void)
{
  M5.begin();

  { /// I2S custom setting
    auto spk_cfg = M5.Speaker.config();

//  spk_cfg.sample_rate =  22050;
    spk_cfg.sample_rate = 1000000;

    spk_cfg.dac_zero_level = 128;
    spk_cfg.pin_data_out = 25;
    spk_cfg.use_dac = true;
    spk_cfg.i2s_port = I2S_NUM_0;

    // if you set `stereo = 1`, you can stereo output with GPIO25 and GPIO26.
    spk_cfg.stereo = 1;

    M5.Speaker.config(spk_cfg);
  }
  M5.Speaker.setVolume(64);
}

void loop(void)
{
  M5.Speaker.tone(1000, 0, 0, false, sin_wav, sizeof(sin_wav));
  M5.Speaker.tone(1001, 0, 1, false, sin_wav, sizeof(sin_wav));
  M5.Speaker.tone(1002, 0, 2, false, sin_wav, sizeof(sin_wav));
}
```

FateEscape, 2022-08-31:
> @FateEscape I do not know what the performance limits of the ESP32's DAC are, but it can operate up to at least 1 MHz.
> 
> BTW, Perhaps you are editing the contents of the library directly? Instead of rewriting the library, you can do the configuration on the user code side.
> 

Yes, i tried editing the config directly and also by user code.
Putting the Sample-Rate up to 1Mhz made the quality of the Sine waves even better.

Of course i'ts not directly comparable with a Waveform Signal Generator, but for getting those things running on a device like the M5 Core2 is truly amazing.

Your work on the Library is very much appreciated, thanks for your help!

closed 2022-08-31 by FateEscape