m5stack/M5Unified #42: ESP-IDF example?

phatpaul, 2022-09-14:
The description says this supports ESP-IDF framework, but I don't see any example or guide to setup in that framework.

BTW I see that https://github.com/m5stack/M5Stack-IDF was archived, so is this supposed to be the new home?

lovyan03, 2022-09-17:
Hello @phatpaul 

Currently there is no sample code available for ESP-IDF, but this library works as a component of ESP-IDF.
Place the `M5Unified` and `M5GFX` folders in the `components` folder of your project.
After that you can use them by putting `#include <M5Unified.h>` in your source files.

Please use ESP-IDF v4.x.x since ESP-IDF v5.x.x is not yet supported.

Since M5Unified is a latecomer, I would like to maintain it as much as possible to solve the dissatisfaction of the earlier libraries.

hp27596, 2022-12-16:
Hello @lovyan03 

Is there a reference ESP-IDF v4.x.x build? I tried building a hello world image using your instructions of placing said folders on the `components` folder, and I got different build errors on v4.1.4 and v4.4.3 on my hello word project. 

```
#include <M5Unified.h>

void setup(){
  M5.begin();
  M5.Lcd.printf("hello world");
}

void loop() {
  M5.update();

}
```

hp27596, 2022-12-16:
When I ran `idf.py build` on v4.4.2 and v4.4.3 these errors are thrown

```
[4/6] Linking CXX executable m5core2test.elf
FAILED: m5core2test.elf
```

And then

```
esp-idf-v4.4.2/components/freertos/port/port_common.c:135: undefined reference to `app_main'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

lovyan03, 2022-12-19:
I apologize for the inconvenience, but examples is configured for ArduinoESP32.

For ESP-IDF, you need to prepare your own app_main function.
The setup function and loop function are not necessary for ESP-IDF because they are Arduino rules.

lovyan03, 2022-12-19:
hello @hp27596 
I apologize for the inconvenience, but examples is configured for ArduinoESP32.

For ESP-IDF, you need to prepare your own app_main function.
The setup function and loop function are not necessary for ESP-IDF because they are Arduino rules.

hp27596, 2022-12-19:
I understand. Pardon my ignorance as there was not a lot of examples or guides to begin with to work with the library.

How would one get started with using the library as a component of ESP IDF? From what I gathered it might look something like this. Can you point us to the right direction?

```
#include <Arduino.h>
#include <M5Unified.h>

extern "C" void app_main()
{
    initArduino();
    M5.begin();
    M5.Display.printf("Hello world!");
}
```

Is that correct?

lovyan03, 2022-12-21:
@hp27596 
I don't see any particular problem with your way of writing.

M5Unified is designed to be ArduinoESP32 library independent.
If the functionality you need is included in ArduinoESP32, you can add it.

hp27596, 2022-12-22:
Hello @lovyan03 

Thanks for your help.

I've found the culprit of the problem. This needs to be in the main folder's `CMakeLists.txt` for ESP IDF to compile. Other than that there's a few quirks that I've worked out too.

`set(SOURCES main.cpp)`

I finally have a working demo which is great. I'm leaving the repo here in case anyone is interested in using the library in bare ESP IDF like I do and want some working examples.

https://github.com/hp27596/m5core2idfdemo

yltsa, 2023-06-09:
Hi, thank you for this great library. Not many vendors support esp-idf.

I could just add a new board into platformio.ini file and the M5Stack AtomS3 worked well with the existing code. The cool stuff is now the small Atom display, which works too.

This is the configuration in platformio.ini:

```
[env:m5stack-atoms3]
platform = espressif32
board = m5stack-atoms3
framework = espidf
build_flags = -DM5STACK_ATOMS3
lib_deps = m5stack/M5Unified@^0.1.6
```

mdxs, 2024-04-17:
This ticket can be closed as resolved; as people indicate it works and are happy with this.