m5stack/M5Unified #43: white noise from SPK HAT

lkebin, 2022-12-02:
Hello, I'm try to test audio on M5StickC Plus with SPK HAT, using [WebRadio_with_ESP8266Audio.ino] example.

How can I avoid the white noise ? or is it expected from design ?

https://user-images.githubusercontent.com/1335935/205261281-046090bb-7b33-44a1-acd0-28f1f6aca4e7.mp4

lovyan03, 2022-12-02:
Hello, @lkebin .

There is a hardware circuit problem, which has been resolved in the newly manufactured ones.
If you have soldering skills, you can reduce noise by connecting a capacitor to GND.

[image]

lkebin, 2022-12-02:
Cool, let me have a try. Thank you !

lkebin, 2022-12-02:
Better than before, Thanks.

closed 2022-12-02 by lkebin