m5stack/M5Unified #45: M5Stack.h and M5Unified.h

ghettokon, 2022-12-26:
Hi, I recently modified a Mini Proto Unit to read two analog sensors (FSR), it works perfectly with M5Stack.h, it works perfectly, both sensors read 0 to 4095 responsively. However, when I switched to M5Unified.h, one of them (GPIO 26) stop working correctly.  Is there a work around to this? Many thanks!

I use a M5Stack Fire.

lovyan03, 2022-12-26:
Hello @ghettokon .
M5Unified and M5GFX change multiple GPIO settings for automatic board recognition during initialization.
Therefore, GPIO settings made before `M5.begin();` may be disabled.
If your code has the necessary settings for analog readout before M5.Begin is executed, try placing them after `M5.begin();`.

ghettokon, 2022-12-26:
Thanks for getting back to me so quickly. My set up is very simple. This is what my setup() looks like this, I combined DUALBUTTONS and ANGEL code together. And in the Loop(), I just do analogRead on both pins. 

When both sensors are untouched, if I use M5Stack.h, both readout are 0. If I use M5Unified.h, GPIO 36 shows 0, but GPIO 26 shows random number between 50 to 150, looks like some kind noise? 

`#include <M5Unified.h>
int sensorPin = 36; 
int sensorPin1 = 26; 

void setup() {
  M5.begin(); //Init M5Stack.  初始化M5Stack
  M5.Power.begin(); //Init power  初始化电源模块
  pinMode(sensorPin, INPUT);  //Sets the specified pin to input mode.  设置指定引脚为输入模式
  pinMode(sensorPin1, INPUT);  //Sets the specified pin to input mode.  设置指定引脚为输入模式
  dacWrite(25, 0);
  M5.Lcd.setTextSize(2);  //Set the font size to 2.  设置字体大小为2
  M5.Lcd.print("Square FSR 1&2: ");
}`

lovyan03, 2022-12-30:
I have tried it and it seems to work fine.

If your problem persists, please provide me with the code in a fully buildable state so that I can test it under the same conditions.

My tested code and results are as follows

```
#include <M5Unified.h>
int sensorPin = 36;
int sensorPin1 = 26;

void setup() {
M5.begin();
pinMode(sensorPin, INPUT);
pinMode(sensorPin1, INPUT);
dacWrite(25, 0);
M5.Lcd.setTextSize(2);
}

void loop(void) {
  auto val = analogRead(sensorPin);
  auto val1 = analogRead(sensorPin1);
  M5.Lcd.setCursor(100,100);
  M5.Lcd.printf("G %d : %d  ", sensorPin, val);
  M5.Lcd.setCursor(100,130);
  M5.Lcd.printf("G %d : %d  ", sensorPin1, val1);
}
```

https://user-images.githubusercontent.com/42724151/210037730-8ae2eb4e-d886-4f14-a987-95f3e05c683b.MP4

closed 2023-11-21 by lovyan03