m5stack/M5Unified #48: Button Events

fbritop, 2023-01-24:
How can we get our own custom callbacks for buttons as we issued to have with Core2?

`          int16_t xx=elem["xx"].as<int16_t>();

          int16_t yy=elem["yy"].as<int16_t>();

          int16_t ww=elem["ww"].as<int16_t>();

          int16_t hh=elem["hh"].as<int16_t>();

          int16_t id=elem["id"].as<int16_t>();

          if(elem["manualOpen"].as<bool>()==true){

            bAcceso = new Button(xx, yy, ww, hh, false, text, onCol, offCol);

            bAcceso->addHandler(eventDisplay, E_RELEASE);

          }else{

            bAcceso = new Button(xx, yy, ww, hh, false, text, onColD, onColD);

          }
          bAcceso->userData=elem["id"].as<int16_t>();`

lovyan03, 2023-01-24:
Hello @fbritop .
Sorry, M5Unified does not currently have that feature.

fbritop, 2023-01-24:
Unified is actually in?
Alpha
Beta
Prototype

Are we hoping that all M5Core2 and other devices current functions will be ported to Unified?. So to know if we will have to move over Unified sometime in the near future

lovyan03, 2023-01-25:
Hello, @fbritop 
Apologies for not including the features you seek.

The main purpose of M5Unified is to absorb the differences between different models and to operate with a common code.
While we would like to provide the same functionality as the existing library as much as possible, it is sometimes difficult to provide it as is.

For example, Support both Arduino and ESP-IDF... Consider how to behave with models that do not have a touch screen...

If you really need it immediately, you can also import the necessary parts from the M5Core2 library and adjust them by yourself.
If M5Unified still seems inadequate, please use M5Core2.h.

fbritop, 2023-01-25:
Thanks, will stick with that. I just was looking for a simple solution for the speaker problem in Core2.h. Maybe I will seek for Speaker functionality in Unified and try to implement in Core2.h