m5stack/M5Unified #50: Missing IMU_Class::getTemp() function declaration

fxstein, 2023-02-03:
`IMU_Class::getTemp()` is defined in `IMU_Class.hpp`, not not implemented in `IMU_Class.cpp` which leads to a link error when attempting to leverage the function.

Here is a possible implementation for `IMU_Class.cpp`:

```
bool IMU_Class::getTemp(float *t) {
  bool res = false;

  switch (_imu) {
    case imu_t::imu_sh200q:
      res = Sh200q.getTemp(t);
      break;
    case imu_t::imu_mpu6050:
    case imu_t::imu_mpu9250:
    case imu_t::imu_mpu6886:
      res = Mpu6886.getTemp(t);
      break;
    case imu_t::imu_unknown:
    default:
      break;
  }

  return res;
}
```
I hope this helps!

PS: Working on an ESPHome wrapper for m5unified.

fxstein, 2023-02-03:
BTW Happy to turn this into a pull request if that helps.

patfelst, 2023-02-03:
> PS: Working on an ESPHome wrapper for m5unified.

Oh I'm very interested in this, if it means I'd be able to use my Core2 with ESPHome!

lovyan03, 2023-02-04:
Hello, @fxstein  .
Thanks for letting me know! It was very helpful.
A new update release will be prepared this weekend.

fxstein, 2023-02-04:
Glad I could help! Looking forward to the new release. Thanks!

lovyan03, 2023-02-06:
Hello, @fxstein 
v0.1.3 has been released.
Please try this Issue as it has been resolved.

fxstein, 2023-02-06:
Confirmed. This is resolved.

closed 2023-02-06 by fxstein