m5stack/M5Unified #51: Wrong week of day from RTC

youkoseki, 2023-02-05:
M5.Rtc.getDateTime() returns wrong week of day since January. I compiled Rtc.ino from the examples on Core2 with M5Unified 0.12 on PlatformIO, and it returns Thursday on Feb 5th (which is Sunday actually). localtime() or gmtime() works without problems, so it looks like only RTC is returning a strange value.

I found the similar report on Twitter as well.
https://twitter.com/hhajime0101/status/1612371508843941889

lovyan03, 2023-02-05:
Hello, @youkoseki .
Thanks for letting us know !
This issue has now been fixed in the develop branch.
https://github.com/m5stack/M5Unified/commit/639e4c46f9b333fb7c0ef56572ff3be63de3af78

An update release will be issued shortly.

lovyan03, 2023-02-06:
Hello, @youkoseki 
v0.1.3 has been released.
Please try this Issue as it has been resolved.

closed 2023-08-12 by youkoseki