m5stack/M5Unified #54: Question about LovyanGFX vs M5GFX

patfelst, 2023-04-17:
Hi Lovyan. I am curious. I seem to recall that M5Unified included LovyanGFX library, however it now looks like it uses M5GFX, which looks like a branch of LovyanGFX. Can I ask why you did this, doesn't it mean you need to duplicate library maintenance? i.e. you need to make the same changes to LovyanGFX and also M5GFX - or do I misunderstand.

Thanks in advance.

lovyan03, 2023-04-17:
Hello, @patfelst
I am the maintainer of both LovyanGFX and M5GFX.
LovyanGFX and M5GFX could evolve separately, but that would probably confuse my poor brain.
Therefore, I would like to keep them as common code as possible.

Incidentally, LovyanGFX is my personal project.
Therefore, if I die, maintenance will end there.
Also, I might lose my motivation and delete the repository abruptly.

On the other hand, M5GFX is an official project of M5Stack.
Therefore, if I die, someone else will take over maintenance.
Also, I am bound by contract and cannot delete the M5GFX repository.

It makes sense to create an M5GFX library that incorporates LovyanGFX as the way the official M5Stack library should be.

patfelst, 2023-04-17:
Thanks @lovyan03 yes I should have said I knew you maintained both libraries. That makes a lot of sense, especially the bit about if you are no longer with us (I hope this isn't the case!).

Have you heard of [ESPHome](https://esphome.io/)? There is a group of us who are looking to create a ESPHome component to work with LovyanGFX (it has just been done with TFT_eSPI and Lilygo T-Display-S3 - but a few of us are keen to now do it with LovyanGFX). But I suggested it might be better to base it on M5Unified. I guess it depends on whether we want to target ESP32s with various LCDs, or whether we want to target M5 Stack products. Do you have any thoughts on this?

Thanks for your time!

lovyan03, 2023-04-17:
Yes, I understand what you are saying.
If I were to express a wish from a user's point of view, I would want a library that crosses company boundaries and supports more products.

One library built from this perspective is tobozo's ESP32-ChimeraCore.
Or perhaps someone could fork M5Unified and create a compatible library that incorporates other companies' products.

Unfortunately, I no longer have the time resources available for its maintenance myself.

patfelst, 2023-04-17:
thank you, that's very helpful.