m5stack/M5Unified #57: Deep sleep on CoreInk

pguyot, 2023-05-09:
Using tag 0.1.6.

Calling `M5.Power.timerSleep(int)` puts a CoreInk to deep sleep, but when it wakes up, if it is running on battery, the reboot reason (result of `esp_reset_reason()`) is not `ESP_RST_DEEPSLEEP` but `ESP_RST_WDT` or `ESP_RST_POWERON` and any RTC Slow memory variable (`RTC_NOINIT_ATTR`) is lost.

The problem doesn't happen if the CoreInk is plugged on USB.
The problem doesn't happen if the following two lines are commented:

```diff
  void Power_Class::timerSleep( int seconds )
  {
-    M5.Rtc.clearIRQ();
-    M5.Rtc.setAlarmIRQ(seconds);
+//    M5.Rtc.clearIRQ();
+//    M5.Rtc.setAlarmIRQ(seconds);
    esp_sleep_enable_timer_wakeup(seconds * 1000000ULL);
    _timerSleep();
  }
```

Also, I haven't been able to get `M5.Power.timerSleep(const rtc_time_t&)` on the CoreInk to work at all.

lovyan03, 2023-05-13:
Thank you for your report...!
I will check and adjust.

patfelst, 2023-08-02:
Hi this may be related, also on CoreInk, when I use:
```c++
  M5.Power.deepSleep(10 * 1000000);
```
it works correctly on USB power, that is, wakes up after 10 seconds. On battery, it goes to sleep but never wakes up. A single push of the power button wakes it up again.

patfelst, 2023-08-04:
Hi guys, I found a solution here https://github.com/m5stack/M5Core-Ink/issues/8

patfelst, 2023-08-05:
thankyou @lovyan03 👍

lovyan03, 2023-08-06:
I was a bit mistaken. It seems @pguyot 's first post contains a misunderstanding of the spec.
The timerSleep is intended to be started using the I2C RTC timer, not a function intended to make it deepSleep.
If you want ESP32 to do deepSleep, use `M5.Power.deepSleep`.

Although the designation of deepSleep is not originally necessary in the timerSleep function, it is included as a fallback action because the ESP32 will not be able to return from timerSleep on models that do not have an RTC.

commit 5986a80e (2023-08-05): fix #57 ( thanks for @pguyot @patfelst )
commit 6cc664c1 (2023-08-05): fix #57 ( thanks for @pguyot @patfelst )
closed 2023-08-06 by lovyan03