m5stack/M5Unified #64: M5CoreS3 and M5Unified repos - unclear relation to CoreS3 support

mhaberler, 2023-06-11:
I see ongoing commits relating to CoreS3 support in both https://github.com/m5stack/M5Unified and https://github.com/m5stack/M5CoreS3, and very recent

can you please clarify which repo is intended to be used for CoreS3 development?

is the intent to fold M5CoreS3 into M5Unified eventually or what is the plan here?

thanks

Michael

lovyan03, 2023-06-11:
Sorry for the confusion.
The respective libraries are not integrated. They are separate.
Both will continue to be supported.

If you want to target only CoreS3, choose `M5CoreS3` library.
If you want to target multiple models with common source code, consider `M5Unified` library.

The M5CoreS3 library is developed to support only M5CoreS3.
In other words, it cannot be used for models other than M5CoreS3.

M5Unified is developed primarily to support multiple products.
Support for multiple models allows for universal use.

Basically, M5Unified's main objective is to be more versatile than a model-specific library.
Therefore, support for model-specific features may be incomplete.

If you are not attracted to integrated libraries, then there is no advantage to using M5Unified.

mhaberler, 2023-06-11:
thanks for the quick reply

I tend to use M5Unified as we will have multiple platforms

a question in principle: assume I hit an issue in M5Unified and it is fixed in M5CoreS3

do you accept PR's to carry over a fix into M5Unified?

best regards

Michael

lovyan03, 2023-06-11:
The main maintainer of each library is a different person.

I was originally in the position of a single user of M5Stack and not a member of M5Stack.
I liked the M5Stack product and used several models, but was frustrated by the lack of consistency in the libraries for each model and the lack of common programming.

I entered into an outsourcing agreement with M5Stack and proposed the M5Unified plan to resolve this situation and began development.

While I welcome pull requests to fix M5Unified issues, I will not actively participate in the work to fix model-specific libraries.

mhaberler, 2023-06-11:
I see

sorry to be an inquisitive bastard.. but I assume many folks have similar questions

I really like the idea underlying the M5Unified effort, and M5GFX is great

we have a mix of various M5stack platforms, including CoreS3, and a few non-M5stack boards like Sunton and Lilygo

my thinking was I could get away with private forks of M5Unified and M5GFX to add support for non-M5 hardware - is that reasonable?

I have  a similar question about M5GFX vs LovyanGFX - which one would be the upstream? I see both repos busy

would I be better off to go with LovyanGFX to cover non-M5 hardware or is the above approach applicable?

I guess you wont accept PR's for M5GFX for non-M5 hardware, but that'd be ok for LovyanGFX, correct?

thanks for your patience

Michael

lovyan03, 2023-06-11:
You may find this Issue helpful.

https://github.com/m5stack/M5Unified/issues/54

mhaberler, 2023-06-11:
thanks, appreciated!

closed 2023-06-11 by mhaberler