m5stack/M5Unified #66: [CoreS3] Ersatz for M5.update() to support M5.BtnX using touch

tobozo, 2023-06-11:
Unlike M5Core2 where the touch surface is bigger than the display, the CoreS3 touch surface is limited to the inner display area.

There are many valid reasons why touch buttons emulation is not activated with CoreS3, but in some projects there is is also a need for a transparent solution.

This snippet enables BtnA, BtnB and BtnC on CoreS3 using modified logic from M5Core2 buttons emulation, without breaking the `Unified` in M5Unified.

```cpp

// 1) optionnally draw three buttons at the bottom of the screen (20px max)
// 2) use `M5Update()` instead of `M5.update()`

void M5Update()
{
  if( M5.getBoard()==lgfx::boards::board_M5StackCoreS3 && M5.Touch.isEnabled() ) {
    // M5Unified doesn't handle Touch->M5.BtnX translation for CoreS3, give it a little help
    uint8_t edge = 220; // spoil 20px out from the touch zone
    auto ms = m5gfx::millis();
    M5.Touch.update(ms);
    uint_fast8_t btn_bits = 0;
    int i = M5.Touch.getCount();
    if( i>0 ) {
      while (--i >= 0) {
        auto raw = M5.Touch.getTouchPointRaw(i);
        if (raw.y > edge) {
          auto det = M5.Touch.getDetail(i);
          if (det.state & m5::touch_state_t::touch) {
            if (M5.BtnA.isPressed()) { btn_bits |= 1 << 0; }
            if (M5.BtnB.isPressed()) { btn_bits |= 1 << 1; }
            if (M5.BtnC.isPressed()) { btn_bits |= 1 << 2; }
            if (btn_bits || !(det.state & m5::touch_state_t::mask_moving)) {
              btn_bits |= 1 << ((raw.x - 2) / 107);
            }
          }
        }
      }
    }
    M5.BtnA.setRawState(ms, btn_bits & 1);
    M5.BtnB.setRawState(ms, btn_bits & 2);
    M5.BtnC.setRawState(ms, btn_bits & 4);
  } else { // trust M5Unified
    M5.update();
  }
}
```

lovyan03, 2023-06-11:
Thanks for the suggestion.

I am also looking to address this issue and am considering implementing a virtual button that would assign the button to any area of the touchscreen.
There are also considerations such as how to handle the effects of setRotation on the display, which should be carefully considered.

tobozo, 2023-06-11:
Indeed every M5Stack device with a Touch interface would benefit from a generic implementation of virtual buttons.

However I do not believe M5CoreS3 needs a specific M5Unified rule for buttons handling as it creates a tech debt for user styles (labels, colours, position), so I am only sharing a sketch solution to a sketch problem I had.

GOB52, 2024-03-16:
[gob_unifiedButton](https://github.com/GOB52/gob_unifiedButton)

Although limited to CoreS3, I implemented rotation support in a library of my own making.  
It is simply rotating.  

I think there is still a lot to take into account in order to generalize.

---
CoreS3 限定ながら、自作のライブラリに回転対応を実装してみました。  
単純に回転させているだけですが。

汎用化にはまだ考慮する点がありそうですね。