m5stack/M5Unified #69: M5Core2 issues when Battery/Screen removed.

NotInTheMafia, 2023-07-28:
I'm having an issue, the M5core2 is the near perfect ready made system for my application.  I have tried to use just the bare board with no screen/touch sensor or battery connected and I have found that the internal I2c bus seams to not be responding.    I see no i2c traffic using a scope after calling M5.begin , subsequent  attempts to read the AXP192 all return 0 the, 5v boost will not turn on...  Is there a quick easy i can disable using the primary display/touch sensor so i can start to narrow this down?
The external I2c Bus still functions/ and the program itself is not hung up.

lovyan03, 2023-07-29:
Hello @NotInTheMafia .
Have you removed the LCD from the main board?
In other words, is the ESP32 unable to communicate with the LCD?
If so, M5Unified will not work.

M5Unified identifies the model by communicating with the LCD at startup.
It is not intended to use M5Unified without the LCD present.

NotInTheMafia, 2023-07-29:
Thank you for your quick response and your helpful clue!!  I think I found the answer.  I was able to simply substitute: 
auto board = _check_boardtype(_primaryDisplay.getBoard());
with this
auto board = _check_boardtype(board_t::board_M5StackCore2); 
5V output is alive again, both i2c busses work, AXP192 reports a voltage...  
So far so good now.  Thank you again!!

closed 2023-11-21 by lovyan03