m5stack/M5Unified #70: CoreInk display fades after some time in use

patfelst, 2023-08-06:
Hi there,

After some time with the display on, something makes it fade. I am using a repeating cycle of wake from deep sleep, clear display, update display with sensor data, wait for user input for 10 seconds, then go to deep sleep for 60 seconds and repeat. After a few hours the ePaper display fades (sorry I don't know exactly how long, but some time).

No amount of my powering down, or resetting, or clearing the display helps unfortunately.

This photo doesn't show it very well, the humidity 41.8% has not faded much, but the battery "Batt: 100%" has faded quite a lot.
[image] Also if you look closely the background still has some grey which hasn't been cleared properly.

I did some searching, and there is some feature called HV on and HV off for the SPI connection to the ePaper display. This issue is discussed on the M5Stack forum [here](https://community.m5stack.com/topic/3158/core-ink-screen-fades-on-reset). Apparently it solves this display fade issue.

I also see the M5Stack library has this HV on and off function for Arduino [here](https://github.com/m5stack/M5Core-Ink/blob/8276d8b95ac75a115f0b542f3a47621d494d61b7/src/utility/Ink_eSPI.cpp#L294) and [here](https://github.com/m5stack/M5Core-Ink/blob/8276d8b95ac75a115f0b542f3a47621d494d61b7/src/utility/Ink_eSPI.cpp#L301).

I would like to try using HV on and off using M5Unified but I don't think those functions are currently available. How hard would it be to implement this? I would be happy to test if you need. Maybe I will try using the M5stack library and see if this solves my problem.

Thank you!

lovyan03, 2023-08-06:
Hello, @patfelst . Thanks for letting me know.
Perhaps executing the following command before going to sleep will solve the problem.

```
M5.Display.powerSaveOn();
```

patfelst, 2023-08-06:
Thanks just tried but it doesn't seem to make any difference, but I think my current problem is I cannot turn off the CoreInk properly to reset the display. As far as I can tell `M5.Power.powerOff()` puts the device in deep sleep (I might be wrong) and pressing reset button on back of CoreInk just, well, resets it.

lovyan03, 2023-08-06:
The `M5.Power.powerOff()` function operates to turn off power by setting the power hold pin LOW.
CoreInk is deepSleep because the power does not turn off while USB power is being supplied due to circuit specifications.
If the USB is not connected, the power should turn off.

patfelst, 2023-08-06:
thanks yes that works to power down the device - ie remove USB power and `.powerOff()`

I'm not sure what fixed it but I did the following and it looks ok again now:
- added `M5.Display.powerSaveOn()` before entering deep sleep
- changed the display rotation
- played with turning the CoreInk power off (I don't think this played a part now)
- changed display mode from `M5.Display.setEpdMode(epd_mode_t::epd_text)` to `M5.Display.setEpdMode(epd_mode_t::epd_fast)`
[image]
[image]

Before I close, what does `M5.Display.powerSaveOn()` actually do?

Thanks for your help!

lovyan03, 2023-08-06:
`M5.Display.powerSaveOn()` should do the same as `powerHVON()` in the CoreInk library.
However, it does not include a delay.
Hence, you may want to wait a few seconds after executing `M5.Display.powerSaveOn()` before turning off the power.

closed 2023-08-06 by patfelst