m5stack/M5Unified #73: M5Stamp C3でM5Unifiedをincludeできない

hashy0917, 2023-09-21:
スケッチのUpload中に以下のようなエラーが発生します

```
c:\Users\■■■■\Documents\Arduino\libraries\M5Unified\src\utility\Power_Class.cpp: In member function 'void m5::Power_Class::_powerOff(bool)':
c:\Users\■■■■\Documents\Arduino\libraries\M5Unified\src\utility\Power_Class.cpp:588:21: error: 'esp_sleep_enable_ext0_wakeup' was not declared in this scope
       if (ESP_OK != esp_sleep_enable_ext0_wakeup( pin, false))
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\Users\■■■■\Documents\Arduino\libraries\M5Unified\src\utility\Power_Class.cpp:588:21: note: suggested alternative: 'esp_sleep_enable_wifi_wakeup'
       if (ESP_OK != esp_sleep_enable_ext0_wakeup( pin, false))
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
                     esp_sleep_enable_wifi_wakeup

Using library M5Unified at version 0.1.9 in folder: C:\Users\■■■■\Documents\Arduino\libraries\M5Unified 
Using library M5GFX at version 0.1.9 in folder: C:\Users\■■■■\Documents\Arduino\libraries\M5GFX 
exit status 1

Compilation error: exit status 1
```

使用デバイスは
`M5Stamp C3` でIDEは `ArduinoIDE 2.2.1`です

lovyan03, 2023-09-21:
@hashy0917 
ご報告ありがとうございます！助かります。
とりいそぎ develop ブランチにて修正を反映致しましたので、よろしければお試しください…！

hashy0917, 2023-09-21:
おお、迅速な対応ありがとうございます。

無事使用する事が出来ました。

closed 2023-10-18 by hashy0917