m5stack/M5Unified #74: build error with esp-idf 5.1.1 for esp32-c3

bhuvanchandra, 2023-09-28:
m5stack/M5Unified/src/M5Unified.cpp:583:20: error: invalid conversion from 'int' to 'i2c_port_t' [-fpermissive]                                                                                                                                                                       
  583 |     In_I2C.setPort(-1, -1, -1);                                                                                                                                  
      |                    ^~                                                                                                                                            
      |                    |                                                                                                                                             
      |                    int                               

possible fix: https://github.com/bhuvanchandra/M5Unified/commit/8e4c4af5496a2b69f6c69e660b3da6bb7b1a6325

lovyan03, 2023-09-28:
@bhuvanchandra 
Thank you for your report !
The develop branch has been updated. This will be reflected in the next release.

lovyan03, 2023-11-23:
I believe this issue has been addressed and resolved in the latest release.
Thanks for your contribution.

closed 2023-11-23 by lovyan03