m5stack/M5Unified #76: Example "Display_Unicode" doesn't compile with M5Unified

maze61, 2023-10-05:
Hello,
first of all cudos to the developer(s) of M5Unified - excellent work!

After modifing `<m5stack path>/src/utility/In_eSPI_Setup.h, At the end of the file add "#define USE_M5_FONT_CREATOR"`, as per README.md, the sketch `"Examples -> M5Stack -> Advanced -> Display -> Display_Unicode"` runs without any problems on Core1 (V2.7) and Core2.

If I'm now "porting" this example to M5Unified by
```
// #include <M5Stack.h>
#include <M5Unified.h>
...
void setup() {
  // M5.begin();        // Init M5Stack.  初始化M5Stack
  auto cfg = M5.config();
  M5.begin(cfg);
```
I got the following compilation error:
```
In file included from C:\ <...> \Display_Unicode\Display_Unicode.ino:20:
C:\ <...> \Display_Unicode\CUF_24px.h:64274:7: error: 'EncodeRange' does not name a type
 const EncodeRange unicode_24pxEncodeRange[] PROGMEM = {
       ^~~~~~~~~~~
C:\ <...> \Display_Unicode\CUF_24px.h:66327:32: error: 'EncodeRange' was not declared in this scope
                               (EncodeRange *)unicode_24pxEncodeRange};
                                ^~~~~~~~~~~
C:\ <...> \Display_Unicode\CUF_24px.h:66327:32: note: suggested alternative:
In file included from c:\Users\Marcus F. Zelezny\OneDrive\Documents\Arduino\libraries\M5GFX\src/lgfx/v1/LGFXBase.hpp:32,
                 from c:\Users\Marcus F. Zelezny\OneDrive\Documents\Arduino\libraries\M5GFX\src/M5GFX.h:21,
                 from c:\Users\Marcus F. Zelezny\OneDrive\Documents\Arduino\libraries\M5Unified\src/M5Unified.hpp:19,
                 from c:\Users\Marcus F. Zelezny\OneDrive\Documents\Arduino\libraries\M5Unified\src/M5Unified.h:5,
                 from C:\ <...> \Display_Unicode\Display_Unicode.ino:18:
c:\Users\Marcus F. Zelezny\OneDrive\Documents\Arduino\libraries\M5GFX\src/lgfx/v1/lgfx_fonts.hpp:126:10: note:   'lgfx::v1::EncodeRange'
   struct EncodeRange {
          ^~~~~~~~~~~
In file included from C:\ <...> \Display_Unicode\Display_Unicode.ino:20:
C:\ <...> \Display_Unicode\CUF_24px.h:66327:45: error: expected primary-expression before ')' token
                               (EncodeRange *)unicode_24pxEncodeRange};
                                             ^
C:\ <...> \Display_Unicode\CUF_24px.h:66327:46: error: expected '}' before 'unicode_24pxEncodeRange'
                               (EncodeRange *)unicode_24pxEncodeRange};
                                              ^~~~~~~~~~~~~~~~~~~~~~~
C:\ <...> \Display_Unicode\CUF_24px.h:66321:30: note: to match this '{'
 const GFXfont unicode_24px = {(uint8_t *)unicode_24pxBitmaps,
                              ^
C:\ <...> \Display_Unicode\CUF_24px.h:66327:46: error: could not convert '{((uint8_t*)((const uint8_t*)(& unicode_24pxBitmaps))), ((GFXglyph*)((const GFXglyph*)(& unicode_24pxGlyphs))), 32, 65439, 25, 4090, <expression error>}' from '<brace-enclosed initializer list>' to 'const GFXfont' {aka 'const lgfx::v1::GFXfont'}
                               (EncodeRange *)unicode_24pxEncodeRange};
                                              ^~~~~~~~~~~~~~~~~~~~~~~
C:\ <...> \Display_Unicode\CUF_24px.h:66327:69: error: expected declaration before '}' token
                               (EncodeRange *)unicode_24pxEncodeRange};
                                                                     ^

exit status 1

Compilation error: 'EncodeRange' does not name a type
```
Which is the same also on a Core2.

As an Austrian, I depend on Unicode (äöüßÄÖÜ°)

Thank you,
Marcus

P.S.: I am looking forward to when `M5.getPin()` will be available.

lovyan03, 2023-10-05:
Thanks for letting me know the problem.
I have confirmed that you are correct, that EncodeRange is in a missing state.

Because the EncodeRange structure is in namespace "m5gfx" ( or lgfx ),
The workaround you can do now is to add `using m5gfx::EncodeRange;`.

```
#include <M5Unified.hpp>  // before.

using m5gfx::EncodeRange; // add this.

#include "CUF_24px.h"    // after.
```

I would consider doing this inside the library if `USE_M5_FONT_CREATOR` is defined.

p.s.
The next update will be released within a week.

maze61, 2023-10-06:
Hello Lovyan03,

thank you for your fast response and also providing a workaround.

Take care
Marcus

maze61, 2023-10-06:
Hello Lovyan03,

examples sketch compiles now on Core 1, Core 2 and Stick C+.
On all three controllers:
`M5.Lcd.drawString("Hello world", 160, 60, 1);` is displayed correctly. 
But for
```
M5.Lcd.drawString("你好  世界", 160, 90, 1);  
M5.Lcd.drawString("Здравствуй  мир", 160, 120, 1);
M5.Lcd.drawString("こんにちは  せかい", 160, 150, 1);
```
only boxes instead of characters are shown
unicodes dingbats via `M5.Lcd.printf("☀☁☂☃☄★☆☇☈☉...";`  are displayed correctly

Thanks, Marcus

lovyan03, 2023-10-07:
It is important to note that M5Unified is **not fully compatible** with the previous library.

Some function arguments may not exhibit the same behavior.
Therefore, some of them must be changed to match M5Unified.

In this example, the fourth argument of `drawString` does not need to be `,1`.

Do the following.

```
M5.Lcd.drawString("你好  世界", 160, 90);  
M5.Lcd.drawString("Здравствуй  мир", 160, 120);
M5.Lcd.drawString("こんにちは  せかい", 160, 150);
```

lovyan03, 2023-10-07:
The conventional library for LCD control is based on a library called TFT_eSPI.
This had the property that the control method switched greatly depending on the font type.
Therefore, it has a special meaning that font number 1 behaves as a GFX font.

M5Unified (M5GFX) has devised a solution to this special situation,
Font number 1 has no special meaning and is simply a missing number.
Therefore, if font number 1 is specified, there is no corresponding font and the default font is used.

maze61, 2023-10-07:
Hello lovyan03,

thanks for your explanation, which improved my understanding of M5 fonts! 

Works now perfectly on both Core1 and Core2!

Please keep your good work on M5Unified going, I know it's a Tantalus task (M5.getPin(), M5.Power(), ...).

Take care,
Marcus

closed 2023-11-21 by lovyan03