m5stack/M5Unified #77: can't ring the speaker of M5Stack CoreS3 with esp-idf

GeorgeNegret, 2023-10-07:
Hi,
I try to ring the speaker of M5Stack CoreS3 on an esp-idf project with M5Unified component , I have no compilation errors but the speaker remains mute , in the arduino framework speaker works fine .

lovyan03, 2023-10-07:
Hello, @GeorgeNegret 
Please give me the version information of the libraries and IDFs you used and the source code you tried.

GeorgeNegret, 2023-10-07:
Hello, @lovyan03 
IDF 5.1 , M5Unified master ,

'#include "M5Unified.h"

void m5_init()
{
   auto cfg = M5.config();
    M5.begin(cfg);
    for (int i = 0; i < 60; ++i)
    {
    	float Hz = 220 * powf(2.0, i / 12.0f);
    	M5.Speaker.tone(Hz, 50, 0, false);
    }
    M5.Display.fillScreen(TFT_ORANGE);
    M5.Display.setCursor(0, 0, 2);
    M5.Display.setTextColor(TFT_BLACK);
    M5.Display.setTextFont(4);
    M5.Display.println("OrangeWheel started !");
}

extern "C" void app_main() {
   	m5_init();
}`

Thanks !

lovyan03, 2023-10-11:
Please give me some time to investigate the cause.

GeorgeNegret, 2023-11-16:
Hi, @lovyan03 
any news about the cause ?
Thanks !

lovyan03, 2023-11-20:
I'm very sorry, I haven't made any progress yet due to frequent additions of travel work and urgent tasks for a while....

lovyan03, 2025-08-27:
Hello, @fariouche 
I'm sorry, but the starter of this issue says, "can't ring the speaker of M5Stack CoreS3 with esp-idf."
It seems your situation is different.
Mixing different situations in the same issue will cause confusion, so please create a new issue, provide the source code you actually tried, and explain so that I can reproduce the same situation.

ainyan03, 2026-08-17:
Thank you for your patience — release 0.2.20 is now out with a series of speaker fixes for ESP-IDF builds (I2S driver lifecycle, clock configuration, sample pair order). Please give it a try.

If the speaker still stays silent on your setup let us know; otherwise I believe this can be closed.

closed 2026-10-30 by lovyan03