m5stack/M5Unified #81: Unit Mini OLED display does not work with M5Stamp-Pico

DAKarp, 2023-10-21:
Using the M5Unified library and the `Displays.ino` example, I've been able to easily get my Unit Mini OLED display to work with an M5 Atom Lite and an M5 Atom S3. But when I try the same display and the same code with my M5Stamp-Pico, the screen is blank.

I've been able to get several other i2c devices to work with my M5Stamp-Pico, so I know the header is soldered properly. Is there a special incantation I need to perform to get this display to work with the stamp?

(I've also tried [this sample code ](https://github.com/m5stack/M5Stack/blob/master/examples/Unit/MiniOLED/MiniOLED.ino) but that also doesn't work.)

lovyan03, 2023-10-21:
Hello. @DAKarp 
Please try this code. (Connect Unit Mini OLED to port A.)
```
#include <M5UnitMiniOLED.h>
#include <M5Unified.h>

void setup(void)
{
  M5.begin();
}

void loop(void)
{
  M5.Display.fillCircle
   ( rand()%M5.Display.width()
   , rand()%M5.Display.height()
   , 8, rand()
  );
}
```
https://github.com/m5stack/M5Unified/assets/42724151/13e42761-a604-4a8b-86c5-9a5985db5e07

DAKarp, 2023-10-21:
Thanks so much! I got it working, and it turns out the issue was the order of imports!

This does not work:
```
#include <M5Unified.h>
#include <M5UnitMiniOLED.h>
```

and this does:
```
#include <M5UnitMiniOLED.h>
#include <M5Unified.h>
```

lovyan03, 2023-10-21:
Yes, that is the intended specification.
If you want M5Unified to control the external display, the header must be included before M5Unified.
If you want to control the external display yourself, the header must be included after M5Unified.

closed 2023-11-21 by lovyan03