m5stack/M5Unified #86: AXP2101_Class.cpp not fully implemented

Hynek-Texl, 2023-10-25:
I'm using M5CoreS3 device with AXP2101 chip. In M5Unified repository looks class `AXP2101_Class.cpp` not fully implemented. Many important methods returns 0. For example:
```
...
  float AXP2101_Class::getBatteryChargeCurrent(void)
  {
return 0;
  }

  float AXP2101_Class::getBatteryDischargeCurrent(void)
  {
return 0;
  }
...
```

In comparison in repository M5CoreS3 exists `AXP2101.cpp`  which is fully implemented. 

Why AXP2101_Class.cpp is not finished?
Is here some possibility that will be `AXP2101_Class.cpp` fully implemented in near future?

Thanks.

lovyan03, 2023-10-25:
Hello, @Hynek-Texl 
I apologize for any inconvenience caused.
The reason these implementations are incomplete is purely due to my lack of development time resources.
I have created classes for AXP2101 based on the AXP192 implementation, but I am starting with the necessary functions first.
As for the unimplemented functions you mentioned, I have to start by checking whether such functions exist in AXP2101, and at this point I do not know whether I can implement them or not.

mhaberler, 2023-11-06:
@Hynek-Texl  if you want to give it a stab yourself, you might find AXP2101 examples around here: https://github.com/lewisxhe/XPowersLib

closed 2026-10-30 by lovyan03