m5stack/M5Unified #87: AtomS3 Button not defined in M5Unified.h

myAlterX, 2023-10-27:
It seems like I can't use the shortcut button implementation on the Unified lib because Btn or BtnA don't seem to be mapping to the proper GPIO in the AtomS3.

Great little device though!

lovyan03, 2023-10-27:
Hello, @myAlterX .
Please show me the version of the library you are using and the source code you have tried.

myAlterX, 2023-10-27:
Hi!

I'm including:

```cpp
#include <M5AtomDisplay.h>
#include <M5Unified.h>
```

then in setup:

```cpp

  auto cfg = M5.config();

  // external display setting. (Pre-include required)
  cfg.external_display.module_display = false;  // default=true. use ModuleDisplay
  cfg.external_display.atom_display   = true;  // default=true. use AtomDisplay
  cfg.external_display.unit_glass     = false; // default=true. use UnitGLASS
  cfg.external_display.unit_glass2    = false; // default=true. use UnitGLASS2
  cfg.external_display.unit_oled      = false; // default=true. use UnitOLED
  cfg.external_display.unit_mini_oled = false; // default=true. use UnitMiniOLED
  cfg.external_display.unit_lcd       = false; // default=true. use UnitLCD
  cfg.external_display.unit_rca       = false; // default=true. use UnitRCA VideoOutput
  cfg.external_display.module_rca     = false; // default=true. use ModuleRCA VideoOutput

  M5.begin(cfg);

```
in loop somewhere I test for:

```cpp
if (M5.BtnA.wasPressed() || M5.BtnB.wasPressed() || M5.BtnC.wasPressed() || M5.BtnEXT.wasPressed())
```

because this doesn't seem to work, I just read the digital pin for now:
```cpp
int val = digitalRead(BTN_PIN);
// Handle button press
if (M5.BtnA.wasPressed() || M5.BtnB.wasPressed() || M5.BtnC.wasPressed() || M5.BtnEXT.wasPressed() || val == LOW)
```
which I have set at the top to 41 thanks to your clever labeling!
```cpp
const int BTN_PIN = GPIO_NUM_41;

I took a peek at the lib and found this comment:
```cpp
/*
  /// List of available buttons:
  M5Stack BASIC/GRAY/GO/FIRE:  BtnA,BtnB,BtnC
  M5Stack Core2:               BtnA,BtnB,BtnC,BtnPWR
  M5Stick C/CPlus:             BtnA,BtnB,     BtnPWR
  M5Stick CoreInk:             BtnA,BtnB,BtnC,BtnPWR,BtnEXT
  M5Paper:                     BtnA,BtnB,BtnC
  M5Station:                   BtnA,BtnB,BtnC,BtnPWR
  M5Tough:                                    BtnPWR
  M5ATOM:                      BtnA
*/
```
Which doesn't seem to have any mention of the AtomS3. Don't know if constants are defined. If they are, it doesn't work for me.

Thanks for your quick reply!

lovyan03, 2023-10-27:
Are you using `M5.update`?
You must run `M5.update` before reading the button.

```
#include <M5Unified.h>
void setup(void)
{
  M5.begin();
}
void loop(void)
{
  M5.delay(1);
  M5.update();  

  if (M5.BtnA.wasPressed()) {
    M5_LOGE("wasPressed");
  }
}
```

myAlterX, 2023-10-27:
No I was not! 
After updating, BtnA does the job.
Thanks very much for your help!
```cpp
  M5.update();
  if (M5.BtnA.wasPressed()) {
    // works
  }
```

closed 2023-10-27 by myAlterX