m5stack/M5Unified #89: IDF 5.1/Arduino 3.0 support:  legacy adc driver migration missing causing a boot crash if an ADC input is used

mhaberler, 2023-11-01:
I observed this:

>Compiling .pio/build/coreS3/lib561/Wire/Wire.cpp.o
In file included from .pio/libdeps/coreS3/M5Unified/src/utility/Power_Class.hpp:17,
                 from .pio/libdeps/coreS3/M5Unified/src/M5Unified.hpp:54,
                 from .pio/libdeps/coreS3/M5Unified/src/M5Unified.h:5,
                 from src/main.cpp:1:
/Users/mah/.platformio/packages/framework-arduinoespressif32/tools/esp32-arduino-libs/esp32s3/include/driver/deprecated/driver/adc.h:19:2: warning: #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively" [-Wcpp]
   19 | #warning "legacy adc driver is deprecated, please migrate to use esp_adc/adc_oneshot.h and esp_adc/adc_continuous.h for oneshot mode and continuous mode drivers respectively"
      |  ^~~~~~~

normally a deprecation warning is not much of an issue

however, since M5Unified uses the legacy adc API and the user code uses an ADC input pin  via Arduino3 , a crash during boot happens:

>abort() was called at PC 0x42030203 on core 0
=> 0x42030203: check_adc_oneshot_driver_conflict at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/deprecated/adc_legacy.c:930 (discriminator 3)

...

the reason is that Power_Class.hpp initiates the legacy adc
then Arduino 3.0 user code initiates the newer esp_adc/adc_oneshot.h (or esp_adc/adc_continuous.h)  API's and you cant have both causing the abort

looks like this code needs to be migrated to the adc_oneshot API similar to https://github.com/espressif/arduino-esp32/pull/7827/commits/74f4582aacfc790ef3c09be8ccb3224cf95ba25c#diff-d2f30d30397442b48494b2372391850aaa539f77977cb1095f5f1d96855598a3 

I've tried fixing the headers and disabling the [legacy code](https://github.com/m5stack/M5Unified/blob/master/src/utility/Power_Class.cpp#L804-L816) and the boot crash goes away confirming my theory

this should take care of both Arduino3 and pure IDF5.1 builds using ADC pins

Michael

ps: I think I remember a remaining warning about ADC migration from I2S code

mhaberler, 2023-11-01:
you can use this example to verify:  https://github.com/mhaberler/arduino3-playground/commit/54b1258a69ed37fdb16b71594010cea9bc1a8ade

lovyan03, 2023-11-23:
@mhaberler 
Thank you for pointing this out.
I am willing to proceed with the work to address this.
I apologize that I am currently overwhelmed with a large number of tasks and will not be able to begin work on this for some time...

mhaberler, 2023-11-23:
no hurry, Arduino 3 is not out yet for good and I can fall back to Arduino 2 for now

ainyan03, 2026-08-17:
Thank you for your patience — an update that should resolve this has just been released as 0.2.20. Please give it a try.

It reworks the battery ADC handling (#312): on Arduino builds it reads through the Arduino-owned ADC driver instead of creating a competing driver instance.

closed 2026-10-30 by lovyan03