m5stack/M5Unified #91: M5Paper wakes too soon from deep sleep when touch wakeup is enabled - with solution

sbesselsen, 2023-11-20:
When I call `M5.Power.deepSleep(10 * 1000000, true)` on the M5Paper, the deep sleep always ends after around 2 seconds with `esp_sleep_get_wakeup_cause() == ESP_SLEEP_WAKEUP_EXT0`. I think this is because the behavior of GPIO pin 36 is not as PowerClass::deepSleep() expects. If I remove the call to `esp_deep_sleep_start()` and instead monitor the state of pin 36 for a while, I see it going from HIGH to LOW after about 2 seconds and then staying there. This is what triggers the ext0 wakeup. I have solved this by creating my own version of deepSleep() with 2 modifications:

* After calling `M5.Display.sleep()` I add `M5.Display.waitDisplay()`.
* Inside the `while (m5gfx::gpio_in(wpin) == false)` loop, I add `M5.update()`, as suggested here: https://community.m5stack.com/topic/5802/m5paper-touchscreen-wakeup-on-gpio-36-constantly-triggering

I can confirm that with these two changes, deepSleep with touch wakeup works as expected, and when I remove either of these two changes, it reverts to its erroneous behavior.

I do notice that all this adds around another 4 seconds before esp_deep_sleep_start() is actually called, so if the wakeup timing needs to be precise, perhaps some adjustment of the remaining micro_seconds would be in order, but that is not a decision for me to make.

Hope this helps!

lovyan03, 2023-12-17:
Hello, @sbesselsen 
I am sorry for the late response. Thank you for giving me this information.
Since the develop branch has been updated, if there are no problems, it will be reflected in the next update release.

mdxs, 2024-04-17:
I assume this has been resolved; there have been several releases since. If so, the ticket can be closed as resolved.