m5stack/M5Unified #93: Minimalist configuration using M5Unified and BluetoothA2DPSink

s1tnk, 2023-12-05:
Thank you for your great software.

I am trying a minimalist configuration using M5Unified and BluetoothA2DPSink. I can play audio with the following code, but the volume control via M5.Speaker does not work (also, tone() does not sound).

I understand that Bluetooth_with_ESP32A2DP overrides several methods of BluetoothA2DPSink and uses its own tri-buffer to handle the data, but I am curious what the minimal code would do.

If possible, could you give me some pointers?

Thanks.

```
#include <M5Unified.h>
#include <driver/i2s.h>
#include <BluetoothA2DPSink.h>

BluetoothA2DPSink A2DPSink;
i2s_pin_config_t PinConfig = {
    .bck_io_num = 12,
    .ws_io_num = 0,
    .data_out_num = 2,
    .data_in_num = I2S_PIN_NO_CHANGE
};

void setup()
{
  M5.begin();
  M5.Speaker.begin();
  A2DPSink.set_pin_config(PinConfig);
  A2DPSink.start("A2DP Test");
}

void loop()
{
  M5.update();
  if (M5.BtnA.wasPressed())
  {
    Serial.println("BtnA was pressed");
    M5.Speaker.setVolume(0);
  }
  else if (M5.BtnB.wasPressed())
  {
    Serial.println("BtnB was pressed");
    M5.Speaker.setVolume(255);
  }
  delay(10);
}
```

s1tnk, 2023-12-05:
Sorry, I forgot to inform you about the environment, we are trying with Core2 and PlatformIO is configured as follows.

```
[env:m5stack-core2]
platform = espressif32
board = m5stack-core2
framework = arduino
lib_deps =
    m5stack/M5Unified
    https://github.com/pschatzmann/ESP32-A2DP
monitor_speed = 115200
monitor_filters = esp32_exception_decoder    
```

lovyan03, 2023-12-05:
Hello, @s1tnk 

…すみません、日本の方だと思いますので、日本語で回答させて頂きます。
（英文の読書きにすごく労力がかかるので…）

M5Unifiedはスピーカーの制御にI2Sを使用していますが、
A2DPSinkのI2S設定を使う場合、I2Sの制御権はA2DPSinkが握っていますから、
M5.Speaker側でのI2Sの操作は出来なくなる、とお考えてください。

M5.Speakerの音量設定は、M5.Speakerが波形を生成する際にのみ効果があります。
つまりM5.Speakerと無関係なライブラリによって鳴っている音には一切効果がありません。

M5.SpeakerとA2DPを併用したい場合は、 examples/Bluetooth_with_ESP32A2DP を参考にしてみてください。

https://github.com/m5stack/M5Unified/blob/master/examples/Advanced/Bluetooth_with_ESP32A2DP/Bluetooth_with_ESP32A2DP.ino

s1tnk, 2023-12-05:
多分そうだろうと思っていたのですが、何か見落としているのではないかと思って質問させて頂いた次第です。
おかげでスッキリしました！！ありがとうございます。

closed 2023-12-05 by s1tnk