m5stack/M5Unified #98: M5CoreS3 speaker issue

SalloCs, 2024-02-05:
Hi. I have tried to play some tones, under 1kHz the output is very noisy. I tried with standard configs too, the result didn't change much.
Can you give me any suggestions?

`void Init()
{
    initArduino();    
    M5.begin();
    M5.Power.begin(); 
    auto spk_cfg = M5.Speaker.config();
    spk_cfg.dma_buf_len = 1024;
    spk_cfg.dma_buf_count = 2;
    spk_cfg.magnification = 1;
    spk_cfg.stereo = true;    

    M5.Speaker.config(spk_cfg);
    M5.Speaker.begin();
   
    M5.Speaker.setVolume(230);
    int x = 0;
    while (x<3)
    {
        M5.Speaker.tone(466, 1000); 
        M5.delay(1000);
        vTaskDelay(2000);
        x++;
    }`

[image]

lovyan03, 2024-02-16:
Hello, @SalloCs 

The speakers installed in the M5Stack are small, so they can easily produce high-pitched sounds, but they don't produce very strong bass sounds.

The tone function is designed to output a waveform that is close to a sine wave, but if you turn the volume up too much, the waveform will saturate and become close to a square wave.
The closer you get to a square wave, the more noticeable the harmonic frequencies will be.
Small speakers naturally have stronger overtone frequencies.

If you want clearer and bass sound, we recommend connecting external speakers using module RCA or similar.

Then, please note the following points:

```
  M5.begin();
//M5.Power.begin();  // ← no need this.
  auto spk_cfg = M5.Speaker.config();
  spk_cfg.dma_buf_len = 256;  // ← 1024 is too big.
  spk_cfg.dma_buf_count = 8;  // ← 2 is too little.
  spk_cfg.stereo = false;     // ← CoreS3 is Mono.
```

closed 2024-02-21 by SalloCs

SalloCs, 2024-02-26:
Thank you for your reply. It seems that the speaker was an issue.