m5stack/M5Unified #100: Cracking sound on speaker using the "M5CoreS3-Esphome/voice-assistant"

vaski666, 2024-02-26:
Hi,
I have successfully set up the voice assistant on a CoreS3 using the Yaml below.
https://github.com/m5stack/M5CoreS3-Esphome/blob/main/voice-assistant/m5stack-cores3.yaml 

The default settings give a lot of "crackling sounds" when the response is given (resposne produced by OpenAI Conversation).
I saw another ticket related to the speaker on the CoreS3 where it seems that the lack of bass causes the same.
The recommended solution there was to add an external speaker, i would if possible prefer to adjust the sound output instead.
Are there any possibilities to do this?

I have a ATOM Echo set up as a voice assistant too and the output on this one does not produce these noises.

lovyan03, 2024-02-28:
Hello, @vaski666 
Sorry, I'm doing maintenance for M5Unified, but not for M5CoreS3-Esphome.
I looked at the implementation of M5CoreS3-Esphome and found that it uses M5Unified to initialize the device, but it doesn't seem to use M5Unified for audio playback.
Therefore, there is nothing I can do.

vaski666, 2024-03-01:
oki thanks i will try to raise the ticket there

closed 2024-03-06 by lovyan03