m5stack/M5Unified #102: M5Core2 Power LED

forstner-m, 2024-03-06:
Hello,

I have tried this function on the M5Core2:
M5.Power.setLed(0);
M5.Power.setLed(1);
M5.Power.setLed(255);

I tried that too:
auto cfg = M5.config();
cfg.led_brightness = 1;
M5.begin(cfg);

Unfortunately, nothing happens with it. There is no problem with the normal M5Core2 library. Therefore, the LED is not broken.

Am I doing something wrong or is the LED not implented?

Thank you very much!

Best regards,
Michael

lovyan03, 2024-03-08:
Hello, @forstner-m 
If the main unit type is Core2 v1.0, you can control the LED, but if it is v1.1, you cannot make detailed adjustments.
This is due to the difference in specifications between AXP192 and AXP2101.

As far as I have checked, it is working in v1.0. What model are you using?

https://github.com/m5stack/M5Unified/assets/42724151/85ef05b8-c6c0-47a4-85d3-9c3551a1be5a

forstner-m, 2024-03-08:
I have a Core2 v1.0 and the led is not switchable. Do you have a code for this function?

forstner-m, 2024-03-09:
Your code works, but this do not work:

#include <Arduino.h>
#include <M5GFX.h>
#include <M5Unified.h>

M5GFX display;

void setup(void)
{
    auto cfg = M5.config();
    cfg.led_brightness = 255;
    M5.begin(cfg);
    display.begin();
}

void loop(void)
{
    while(1)
    {
        M5.delay(1000);
        M5.Power.setLed(0);
        M5.delay(1000);
        M5.Power.setLed(255);
    }
}

If the line "display.begin();" is active the LED is off.

lovyan03, 2024-03-09:
M5Unified includes M5GFX.
M5.Display is an instance of M5GFX.
You must not create your own instances of M5GFX.
Examples of such code are sometimes introduced, but they should be avoided as they can cause problems.

forstner-m, 2024-03-09:
Now it works! Is the line M5.Display.begin(); necessary?
It would still be good if an example were created with canvas. But with this line it works: M5Canvas canvas(&M5.Display);

mdxs, 2024-04-17:
This ticket can be closed as resolved.