m5stack/M5Unified #104: M5Capsule cannot be detected corretly

peterchenyipu, 2024-03-08:
Library versions:
- M5Unified 0.1.13
- M5GFX 0.1.13
- M5Capsule 1.0.0

Issue:
None of the examples in M5Capsule would work due to the fact that the board type is not detected correctly. I confirmed with `M5.getBoard()`, and the result is 137 (board_M5AtomS3Lite). The code that resulted this is here:
https://github.com/m5stack/M5Unified/blob/725c654ea1af75e7b80f84723ed612796d40af87/src/M5Unified.cpp#L489-L498
The result variable is 54 or 0b110110. Taking the lowest 4 bits gives 0b0110 or 6. This corresponds to board_unknown in row 2 column 3. As a hack I modified it to be board_M5StampS3 and the examples started working again.

Is this a bug in the library or is there something wrong with my device? I only have one M5Capsule so I cannot verify if the other devices also have this issue. Thank you!

lovyan03, 2024-03-08:
Hello @peterchenyipu .

Thank you for reporting this issue.
I overlooked this issue when I did the last update.

I have now updated the develop branch and changed it so that if the lower 2 bits of the bit pattern are 2, it is considered a Capsule. I think this will solve the problem.

Distinguishing between AtomS3Lite and StampS3 is quite a difficult problem, and I'm not sure how to tell the difference exactly. I am choosing an implementation that seems to have a high probability based on the results of my experiments, but I would welcome any suggestions for other better methods.

peterchenyipu, 2024-03-08:
Thank you for the update!

closed 2024-03-08 by peterchenyipu