m5stack/M5Unified #110: changing font

7h30th3r0n3, 2024-05-10:
Hi ! thanks for the great work ! 

What is the way to change the font ? 

I tried things like :
`M5.Display.setFont(&fonts::Font4);`
or
```
void gfxSetup (LGFX Device* gfx) {
 gfx->setFont(&fonts::efontCN 12);
}
```
but it seem to not changing any font at all, and chinese characters are not printed correctly on screen while using 
`M5.Display.println("something);`

more info : https://github.com/7h30th3r0n3/Evil-M5Core2/issues/2

lovyan03, 2024-05-10:
Hello, @7h30th3r0n3 .
Please try the code below.

```cpp
#include <M5Unified.h>

void setup() {
    M5.begin();
    M5.Display.setFont(&fonts::efontCN_12);
    M5.Display.print("你好世界");
}
```

[image]

I looked at the code in the URL you linked, and it seems likely that you are using the setTextFont function to disable the font each time.
After setting the font, we recommend checking whether the font has been changed unintentionally before drawing

7h30th3r0n3, 2024-07-31:
Thanks for your time 😉

closed 2024-07-31 by 7h30th3r0n3