m5stack/M5Unified #119: M5Unified and Arduino 3.0 on Core2

nuno636, 2024-07-25:
Hello,
after running a few tests I realized that the M5Unified library is not compatible with Arduino 3.0 and IDF 5.1
I took the display example (https://github.com/m5stack/M5Unified/tree/master/examples/Basic/Displays), run it **WITHOUT** Arduino as a component and it works perfectly.
I **then** added Arduino and it stopped working:
Backtrace: 0x400d73f5:0x3ffb92f0 0x400e12e2:0x3ffb9360 0x40089ada:0x3ffb9380
0x400d73f5: lgfx::v1::LGFXBase::width() const at /esp/test/components/M5GFX/src/lgfx/v1/LGFXBase.hpp:311
 (inlined by) setup() at /esp/test/main/test.cpp:233
0x400e12e2: loopTask(void*) at /esp/test/components/arduino/cores/esp32/main.cpp:59
0x40089ada: vPortTaskWrapper at /esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

I tried to disable as many Arduino libraries as possible (SPI, Ethernet, SD, SPIFFS, OTA, WifiProv and so on) but the result is the same. Any operation on the display leads to a crash.

Does anybody have the same problem or any idea to where to start to inspect this problem?

Thank you

mhaberler, 2024-07-27:
same as https://github.com/m5stack/M5Unified/issues/89

currently M5Unified is Arduino2 AFAICT

nuno636, 2024-07-29:
I wouldn't say it's the same as #89 and #90, maybe it's another issue to be put on the same line. At least the other 2 issues throw an error (and the #89 is quite easy to fix on a Core2, not sure about other devices). Mine shows absolutely nothing.

Truth be told, I was fine with the Core2 library and IDF 3.4.x. I just wanted to test the latest library with the latest IDF but I see it's a bit too early. 

I'd gladly fix it, if only I could find where to start from.

gzah3053, 2025-03-02:
I have been trying for some time to get ESP-IDF to work with arduino and M5Unified and other libraries. Which versions are compatible / recommended?

I am using an M5 Core 2.