m5stack/M5Unified #125: AXP2101 Voltage for DLDO1 (LCD BL) with M5Stack CoreS3

lboue, 2024-09-23:
Hello,

I'm trying to find out what voltage to apply to the CoreS3 LCD backlight?

Could you tell me the voltage for DLDO1 (LCD BL)?

https://github.com/m5stack/M5Unified?tab=readme-ov-file#axp2101-usage

[image]

Regards

felmue, 2024-09-23:
Hello @lboue 

I think LCD brightness is handled [here](https://github.com/m5stack/M5GFX/blob/master/src/M5GFX.cpp#L403).

You don't need to care about the actual voltage - just use setBrightness() function.

Thanks
Felix

lboue, 2024-09-24:
Thans @felmue.
I'd like to understand how it's managed because I need it to implement it in Meshtastic.

felmue, 2024-09-24:
Hello @lboue 

well, the AXP2101 datasheet states that the voltage on DLDO1 is the value written to register 0x99 plus an offset of 0.5 V. From the datasheet:
```
dldo1 output voltage configuration
0.5~3.4V，100mV/step，29steps
00000: 0.5V
00001: 0.6V
……
11100: 3.3V
11101~11111: reserved
```
The function setBrightness(brightness) takes an 8 bit value so the range is 0 to 255. A value of 0 turns DLDO1 off via register 0x90. A value from 1 to 255 turns DLDO1 on (reg 0x90) and sets a voltage of 0.6 V to 3.3 V (using register 0x99) via this calculation: `brightness = ((brightness + 641) >> 5);`

So for the max value 255 the value written to 0x99 is: `((255 + 641) / 32) = 28` (equals 0b00011100)

BTW: Isn't Meshtastic written in C as well? Maybe you could just copy the complete function over?

Thanks
Felix

lboue, 2024-09-24:
Thanks for your help.

closed 2024-09-24 by lboue