m5stack/M5Unified #127: M5StickC Plus 2 speaker is crashing program

vectorcrumb, 2024-09-29:
Hello, I'm trying to use the Speaker on a M5StickC PLUS2. I'm using Platform.IO with the esp and arduino frameworks. The relevant setup code here is:

```cpp
auto m5_cfg = M5.config();
m5_cfg.external_rtc = true;
m5_cfg.internal_spk = true;
M5.begin(m5_cfg);
M5.Speaker.setVolume(200);
```

I then call the speaker with:

```cpp
M5.Speaker.tone(2000, 1000, 0, true);
```
This causes a crash. I've attached the relevant parts of the coredump below

```
       TCB             NAME PRIO C/B  STACK USED/FREE
---------- ---------------- -------- ----------------
0x3ffdd170         spk_task      2/2         2768/468
0x3ffc21b4             main      2/1       2544/14336
0x3ffd72ec             RFID      1/1        1120/7068
0x3ffc2914             IDLE      0/0         416/1104
0x3ffc3074             IDLE      0/0         416/1104
0x3ffde784              LCD      2/2         672/7508
0x3ffc7340              tiT    18/18         544/3028
0x3ffdec98              WDG      3/3         432/3648
0x3ffcf594        mqtt_task      5/5        880/15500
0x3ffaf788             ipc0    24/24         448/1084
0x3ffbc968             ipc1    24/24         448/1076
0x3ffc9524   arduino_events    19/19         432/3648
0x3ffc83c4          sys_evt    20/20         528/2272
0x3ffbde24        esp_timer    22/22         432/3648
0x3ffcbfc0             wifi    23/23         496/6148
0x3ffc39d4          Tmr Svc      1/1         448/1584

==================== THREAD 1 (TCB: 0x3ffdd170, name: 'spk_task') =====================
#0  0x400835a5 in panic_abort (details=0x3ffe846c "***ERROR*** A stack overflow in task spk_task has been detected.") at /home/lucas/.platformio/packages/framework-espidf@3.40406.240122/components/esp_system/panic.c:408
#1  0x4008ad78 in esp_system_abort (details=0x3ffe846c "***ERROR*** A stack overflow in task spk_task has been detected.") at /home/lucas/.platformio/packages/framework-espidf@3.40406.240122/components/esp_system/esp_system.c:137
#2  0x4008ece9 in vApplicationStackOverflowHook (xTask=<optimized out>, pcTaskName=0x3ffdd1a8 "spk_task") at /home/lucas/.platformio/packages/framework-espidf@3.40406.240122/components/freertos/port/xtensa/port.c:407
#3  0x4008ce8d in vTaskSwitchContext () at /home/lucas/.platformio/packages/framework-espidf@3.40406.240122/components/hal/esp32/include/hal/cpu_ll.h:39
#4  0x4008ae73 in _frxt_dispatch () at /home/lucas/.platformio/packages/framework-espidf@3.40406.240122/components/freertos/port/xtensa/portasm.S:436
```

For some reason, the speaker task is crashing due to an overflow in the stack size of the task. Any ideas what could be going wrong in my config or might this be a bug?

felmue, 2024-10-04:
Hello @vectorcrumb 

your example code runs fine for me - no crash.

Are you using the latest versions of `M5Unified`, `M5GFX` and `M5StickCPlus2` libraries?

Thanks
Felix

lovyan03, 2024-11-05:
Hello, @vectorcrumb 
Sorry for the late reply.

I am unable to reproduce the crash.
Please try the latest develop branch.
If you're still experiencing crashes, please provide us with the minimum source code that builds completely.