m5stack/M5Unified #128: Questions about M5.BtnA.releasedFor and M5.BtnA.wasReleaseFor

patfelst, 2024-10-07:
Hi there, I'm using an M5AtomS3. I need to execute some code only once, about 1500ms after BtnA has been released after being held for some time. When I use `if (M5.BtnA.releasedFor(1500))`, it does become true 1500ms after being released, but then remains true forever so the code is also executed forever.

Also if I try ` if (M5.BtnA.wasReleaseFor(1500))` it never seems to be true. Is there a way to reset this button status, or do I need to add my own boolean flag to keep track of when the code has been executed so that it only happens once?

Here is my code:
```c++
  // Display the LED brightness while the user adjusts it (or finishes adjusting) in the main loop
  if (M5.BtnA.isHolding() || M5.BtnA.wasReleasedAfterHold()) {
    // Label
    M5.Display.setTextPadding(0);
    M5.Display.setTextColor(TFT_WHITE, TFT_BLACK);
    M5.Display.setTextDatum(top_left);
    M5.Display.drawString("LED Brt:", 0, y_pos);

    // Value
    M5.Display.setTextPadding(40);
    M5.Display.setTextColor(TFT_GREEN, TFT_DARKGREY);
    M5.Display.setTextDatum(top_right);
    M5.Display.drawNumber(led_bright, wd - 5, y_pos);
  }
  else if (M5.BtnA.releasedFor(1500)) {
    M5.Display.clear(); // <== Keeps executing over and over 1500ms after user releases button
  }
```
The following code works but is not as neat as using built in M5 functions
```c++
  if (M5.BtnA.isHolding() || M5.BtnA.wasReleasedAfterHold()) {
    relased_ms = millis();
    bright_txt_cleared = false;
    // Label
    ...
    // Value
    ...
  }
  // Display the LED brightness while the user adjusts it (or finishes adjusting) in the main loop
  else if (!bright_txt_cleared && millis() - relased_ms > 1500) {
    bright_txt_cleared = true;
    M5.Display.clear(); // <== Executes only once 1500ms after user releases button
  }
```
thanks for any suggestions!

lovyan03, 2024-11-05:
Hello, @patfelst 
Sorry for the late reply.
Are you running the `M5.update();` function properly?
Calling this function will update the state of all buttons, so you must call it once before reading the button state. If you do not call it or call it more than once, the state will not be updated correctly.

patfelst, 2024-11-08:
thanks, I will check this soon!

patfelst, 2024-12-28:
apologies for the long delay in getting back to you.

You are correct, I wasn't calling M5.update() properly, I had to insert another M5.update() before checking the button state, and now the M5.BtnA.releasedFor(1500)) works ok.

Thanks very much.

closed 2024-12-28 by patfelst