m5stack/M5Unified #129: AtomS3R IMU

patfelst, 2024-10-10:
Hi there, I tried the IMU example on a AtomS3R but it says the IMU is not found. 

Are there any plans to update the library to work with the AtomS3R's BMM150 geomagnetic sensor and BMI270 attitude sensor?

Thankyou!

lovyan03, 2024-10-11:
Hello, @patfelst 
Sorry, the current release version doesn't work properly.
Please try the develop branch version.

patfelst, 2024-10-11:
geat thank you, it's working now!

patfelst, 2024-10-11:
Sorry one more question, it looks like a function was going to be implemented to calculate roll, pitch and yaw (heading), otherwise known as AHRS. Do you plan on implementing this
https://github.com/m5stack/M5Unified/blob/77017d17b915f64587b4ddcbfa898547573cedfb/src/utility/IMU_Class.hpp#L116

Or should I use some library like https://github.com/arduino-libraries/MadgwickAHRS?

Thanks again

lovyan03, 2024-11-05:
Sorry for the late reply.

Thank you for providing important information.
I overlooked that the orientation of the IMU of the AtomS3R series is different from the conventional model. I have updated the develop branch now and changed the axis orientation.
If you want to try it immediately, please use the latest develop branch for both M5GFX and M5Unified.

Also, if you want to call the axis setting function, do it as follows.

```
  M5.Imu.setAxisOrderLeftHanded(m5::IMU_Class::axis_t::axis_x_pos, m5::IMU_Class::axis_t::axis_y_pos);
```

patfelst, 2024-11-08:
I will test this very soon, thank you

patfelst, 2024-12-28:
hi again apologies for the long delay.

Is the new AtomS3R orientation code now in the latest release v0.2.2? The orentation seems the same to me. If I use this, the orientation is correct (i.e. heading increases when AtomS3R is turned clockwise).
`M5.Imu.setAxisOrderLeftHanded(m5::IMU_Class::axis_t::axis_x_pos, m5::IMU_Class::axis_t::axis_y_pos);`

I think we can probably close this issue now.

lovyan03, 2025-01-27:
In the currently released version, the IMU orientation of AtomS3 and AtomS3R has been adjusted to be the same.
As far as I understand, there are no compatibility issues at this stage.
If you are having any issues, please let me know the details.

patfelst, 2025-01-27:
I'm no longer working on this project. But like I saide the function setAxisOrderLeftHanded worked for me.

closed 2025-01-27 by patfelst